wled/WLED #2149: Icicles - Set length of each drop to allow dripping etc.

mekenthompson, 2021-08-21:
**Is your feature request related to a problem? Please describe.**
New feature

**Describe the solution you'd like**
Ability to define sections of LED strip that fork off a main trunk of LEDs, such as Icicles. Then be able to implement effects such as dripping icicles / having forks in sync etc.

Thank you for your ideas for making WLED better!

blazoncek, 2021-08-21:
You can do that with segments.

mekenthompson, 2021-08-22:
> You can do that with segments.

I can't achieve what I want with segments. In current implementation there is a maximum of 15 segments. In order for segments to work in my use cased I'd need:

1. No limit on segment numbers; in my current run I'd need 90 segments for each icicle drop, 360 LEDs in 3-4-5 repeating layout.
2. Rather than add each segment via GUI, Ideally be able to this via a config file and/or be able to specify a segment pattern...ie. mine, segments broken into 3-4-5 nodes repeating to 360 total nodes.

blazoncek, 2021-08-22:
Just because you need 90 of them doesn't mean you cannot do it with segments.
You can use multiple ESP32 where each supports 24 segments.
90 segments is too many even for ESP32. So it may never be implemented the way you described it.

mekenthompson, 2021-08-22:
Is 24 the max limit of segments supported by ESP32?

I could look at multiple controllers if needed, was hoping to solve without though

blazoncek, 2021-08-22:
If you want to maximize number of segments you could use my fork (dev branch) where the limit is 32 for ESP32 and 18 for ESP8266. Beyond that resources get depleted pretty soon and WLED becomes unstable.
When you have multiple devices set up, use Sync functionality to keep them synchronized.

What you can do, if you want to use only one ESP, is develop a usermod where you would take control of LED output and use built-in functions for effects and such. I am not saying this is easy, though. ;)

blazoncek, 2021-08-24:
If you have no more comments or requests I am going to close this issue since the functionality _can_ be achieved by creating segments. The maximum number of segments unfortunately cannot be safely increased beyond current amount.
Feel free to reopen if needed.

Another hint/suggestion I can assert is to make a custom effect if your _icycles_ can be programmatically defined.

closed 2021-08-24 by blazoncek