wled/WLED #2151: WiFi connection by BSSID

Aircoookie, 2021-08-21:
**Is your feature request related to a problem? Please describe.**
WLED tends to connect to the incorrect AP in mesh networks (multiple APs with the same credentials)

**Describe the solution you'd like**
Allow optionally specifying a WiFi BSSID (AP mac address - 6 bytes in format 01:23:45:67:89:AB). WLED will only connect to that BSSID. `WiFi.begin()` accepts the BSSID as an optional parameter.

**Describe alternatives you've considered**
Mesh systems might have configuration options to deal with this, but that depends on the manufacturer.

Wireheadbe, 2024-01-20:
This would be a great addition. For a specific WLED, I have a far-away AP on channel 1, while the one in the same room as the WLED, on channel 6, never gets chosen. Forcing a roam, just causes the WLED to disconnect from WiFi and never come back.

ecov773, 2024-01-26:
I have the same issue as @Wireheadbe. Would be nice to be able to choose the BSSID.

blazoncek, 2024-01-26:
I may add such feature into `multiwifi` branch. But no promises made.

blazoncek, 2024-04-06:
No.

PR #4429 (2025-02-19): Add ability to enter desired BSSID
- add event handling (for debug purposes and correct status setting)
- fixes #2151
- moved `WLED::initEthernet()` and `WLED::findWiFi()` to `network.cpp`
- added string conversion functions for MAC address (without colons)
- used MAC conversion functions for configuration
- replaced `WLED_WIFI_CONFIGURED` with function call (and updated macro)
- fixed correct use of `WLED_ETHERNET`
- added AP client counter (not really used currently but it will get its meaning soon if you stop bugging me with updating PR descriptions) 😉 

Oh, and BTW, there is a bug in `src/ [... cut]
commit 272129f6 (2025-02-19): Add ability to enter desired BSSID
- add event handling (debug)
- fixes #2151
closed 2025-02-19 by blazoncek