wled/WLED #2153: WLED odd flickering

FigJam23, 2021-08-23:
Please help!!!! Any suggestions or ideas.....

Hey guys this my first WLED setup using RGBW lefs with a esp32 5v and capacitor unfortunately I don't have a level shifter yet as you can see in my video below I'm getting lots of flickering and color changes while set to low brightness white. The flickering happens across all colors and modes. Would this be because of the level shifter not being used or potentially a Fw issue like above post might suggest.
If I do try a different version .bin what should I try and is there a way to backup my settings and restore too new fw .
Thanks in advanced

https://photos.app.goo.gl/gtNWr8DhYRtnnNyT6

blazoncek, 2021-08-23:
Join WLED Discord for support questions.

Aircoookie, 2021-08-23:
Please don't copy paste in multiple threads, closing this in favor of your comment to https://github.com/Aircoookie/WLED/issues/2137

If the flickering gets worse at low brightness, you almost definitely need a levelshifter or "sacrificial pixel" (first LED with a diode to drop its supply voltage to 4.0V)

closed 2021-08-23 by Aircoookie