wled/WLED #2155: Defaults Turn LEDs on after power up/reset Check box is invalid

neckyaan, 2021-08-24:
**Describe the bug**
WLED 0.13.0-b0 (build 2106302)  Defaults Turn LEDs on after power up/reset Check box is invalid

neckyaan, 2021-08-24:
> **Describe the bug**
> WLED 0.13.0-b0 (build 2106302) Defaults Turn LEDs on after power up/reset Check box is invalid

thank you for all your hard work and devotion you have put into this Aircookie.

Aircoookie, 2021-08-24:
Hi and thank you for the report!
What do you mean with the checkbox being invalid?
Does the browser display an issue or is the option not doing what you expect?

neckyaan, 2021-08-25:
Yes, a preset is selected. But WLED 0.12, WLED 0.13.0-b2 have no such problem, the same setting.

blazoncek, 2021-12-18:
Boot options slightly changed in recent beta.
Closing.

closed 2021-12-18 by blazoncek