wled/WLED #2163: Install in boat, ws2811 (no wifi)

otisrhw, 2021-08-29:
I'd like to install addressable leds in my boat (12v), what would you recommend? Should I get a travel router and create a network? Can I connect to D1 on hotspot (I've tried no luck)? Can I use wifi direct? 
 I have cellular network, Bluetooth and could get travel router to connect to hotspot.

blazoncek, 2021-08-29:
You ca use WLED in AP mode. It does work, though some people reported some issues after a prolonged use, requiring WLED reboot to fix.

If you are not getting WLED AP then your ESP does not boot properly when connected to boat's PSU. You may need to diagnose that using electrical tools.

fusionstream, 2022-02-05:
Maybe time to close this?

closed 2022-02-05 by blazoncek