wled/WLED #2164: ESP32 Board Resetting with WLED

spartandrew18, 2021-08-29:
**Describe the bug**
This is a bug with the board resetting often. Trying to find out the cause.

**To Reproduce**
Plug in the esp32 board into USB and power up the serial monitor with Arduino IDE. Wait for a reset to occur.

**Expected behavior**
I expected to receive no errors. However, it seems to happen on 2 brand new boards.

**WLED version**
 - Board: [ESP32 dev]
[image]
[image]
 - Version [WLED 0.12.0 (build 2104020)]
 - Format [https://install.wled.me/]

**Additional context**
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4016b537  PS      : 0x00060d30  A0      : 0x80101ef5  A1      : 0x3ffb5620
A2      : 0x00000000  A3      : 0x3ffd6a21  A4      : 0x00000032  A5      : 0x3ffd6a22
A6      : 0x0000007d  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00fb56f0
A10     : 0x00000001  A11     : 0x00000002  A12     : 0x3ffc89fc  A13     : 0x77250f76
A14     : 0x00000024  A15     : 0x3ffd69cc  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000034  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x4016b537:0x3ffb5620 0x40101ef2:0x3ffb5650 0x400ebe82:0x3ffb5690 0x400ebf0b:0x3ffb56f0 0x4019781d:0x3ffb5710 0x4010209a:0x3ffb5730 0x401020e3:0x3ffb5760 0x40102105:0x3ffb5780 0x40167b42:0x3ffb57a0 0x401686c6:0x3ffb57c0 0x4016ebab:0x3ffb57e0 0x40173476:0x3ffb5810 0x401605f7:0x3ffb5830 0x4008b0a2:0x3ffb5860

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Ada

--------- I THEN CLICKED THE RESET BUTTON ON ESP 32 BOARD -----------

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
⸮Ada

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Ada

Thank you for your help!

sansillusion, 2021-08-30:
In your log there is a ...
Brownout detector was triggered

Your board does not receive enough power. Try a different usb cable and/or
power sources. You may want to power your board from the Vin or if you have
a good, stable 3.3V power source apply power directly to 3.3v pin.

davea99, 2021-09-10:
Is this with anything else connected to the board or just USB? I've been having similar problem with the board in the setup, tried all sorts with the power supply thinking that might be the problem, added my own 3.3v regulator - turned out to be missing a resistor on the relay control wire which must have been drawing too much current on the GPIO!

mase76, 2021-10-02:
With some of my PCBs I have brownout problems when doing OTA updates. The ESP32 Wroom is powered by a AMS1117 3,3V regulator, connected to a terminal block. I tried different power supplies.

reinethernal, 2022-02-19:
same board, same problem same error(((

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff000c,len:0
ho 12 tail 0 room 4
load:0x00000000,len:0
load:0x00000000,len:-1073741824
ets Jun  8 2016 00:22:57

blazoncek, 2022-02-21:
WatchDog Reset is common when code does not yield CPU for maintenance tasks.
I have seen those happen on incorrect/bad flash procedure.
Please erase flash and try flashing web flasher or esptool. If it produces the same reboot, try uploading debug build (use -D WLED_DEBUG in PIO) using VSC + PIO.

softhack007, 2023-01-12:
If I understand correctly, this problem might be hardware / wiring related ( brownout, and drawing too much current out of a GPIO pin). 
Not enough information to reproduce, and no activity since 11 month -> closing.

closed 2023-01-12 by softhack007