wled/WLED #2165: WLED on another languages

04aromeo, 2021-08-29:
I'm Spannish, sorry for the translation.

Hi, I'm thinking about translating the app to different languages. I could translate it to Spanish, French and Vasque, but I need to know how to edit the app texts. Can anyone tell me how to edit? Thanks.

blazoncek, 2021-08-30:
All of the strings are in JS and HTML files inside data folder.
Some strings may be in usermods.

neckyaan, 2021-08-30:
Please refer to：
https://github.com/Aircoookie/WLED/wiki/Add-own-functionality

04aromeo, 2021-08-30:
> All of the strings are in JS and HTML files inside data folder.
> Some strings may be in usermods.

Oh, thanks, I'm going to try it

04aromeo, 2021-08-30:
> Please refer to：
> https://github.com/Aircoookie/WLED/wiki/Add-own-functionality

Ohh, that's exactly what I was finding about. Thank you very much! 👍 :)

Aircoookie, 2022-10-21:
Closing as resolved and duplicate of #459

closed 2022-10-21 by Aircoookie