wled/WLED #2169: Platform.io env d1_mini_debug does not compile

push-pop, 2021-09-02:
**Describe the bug**
The env d1_mini_debug does not compile with VS code and Platform.io

**To Reproduce**
Get latest branch. Uncomment 
default_envs = d1_mini_debug
select Default->General->Build All

**Expected behavior**
Expect the firmware to build in debug mode

**WLED version**
 - Board: ESP2866 NodeMCU clone
 - Version: 0.13.0-b2
 - Format: Self Compiled

**Additional context**
Error in Build Output:
`
Linking .pio\build\d1_mini_debug\firmware.elf
c:/users/turle/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\d1_mini_debug\lib557\FastLED\lib8tion.cpp.o: file not recognized: file truncated
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\d1_mini_debug\firmware.elf] Error 1
`

Thank you for your help!

push-pop, 2021-09-02:
Oddly enough, if I just copy the settings from d1_mini_debug over to d1_mini that builds just fine and debug works. Not sure what else could be the culprit...

blazoncek, 2021-09-03:
This is not a WLED error but PIO + VSC quirk.
You will need to fix your PIO setup.

closed 2021-09-07 by blazoncek