wled/WLED #2171: Support for ESP32-SOLO-1?

ChuckMash, 2021-09-03:
Hi,

I've recently been tearing into a cheap WiFi RGBWW analog strip and find that it is running on a ESP32-SOLO-1.
A Single core ESP32.
[image]

I've successfully flashed with tasmota32solo1 and have found full control color and white channel control is possible.

As expected, attempts to flash with WLED esp32 binaries do not work.
`
E (502) cpu_start: Running on single core chip, but application is built with dual core support.
E (502) cpu_start: Please enable CONFIG_FREERTOS_UNICORE option in menuconfig.
`

However, if we did get it working, it should be able to set WLED to PWM RGBWC and selecting the corresponding pins on this pre-fab board. 

I'm happy to provide more details, but this is the bulk of what I've figured out so far.
Thoughts?

blazoncek, 2021-09-03:
Look at the PlatformIO docs if the SOLO is supported by it (most likely), create a build environment based on docs specifications, compile for the environment and test.
When you have it working, create a PR with the setup you used.

Other alternative is to provide us with a sample unit and wait until we figure it out. 😄

ChuckMash, 2021-09-06:
I was able to use [this]( https://github.com/lbernstone/arduino-esp32-solo) to (apparently) drop in support for ESP32-Solo-1
and compile and flash WLED for the Solo board.

The WLED Access Point comes online and the captive portal seems to work at first, however nothing comes back for 4.3.2.1.
Examining the serial logs I see that it goes into a boot loop 

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Ada
E (9149) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (9149) task_wdt:  - IDLE0 (CPU 0)
E (9149) task_wdt: Tasks currently running:
E (9149) task_wdt: CPU 0: wifi
E (9149) task_wdt: Aborting.
abort() was called at PC 0x4017ef08 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008835b:0x3ffbfbb0 0x400885c1:0x3ffbfbd0 0x4017ef08:0x3ffbfbf0 0x40086da2:0x3ffbfc10 0x4008aba0:0x3ffafa80 0x40089a24:0x3ffafaa0 0x4017ec06:0x3ffafae0 0x4011ffb1:0x3ffafb00 0x40149372:0x3ffafb20 0x4008a046:0x3ffafb50

Rebooting...
```

ChuckMash, 2021-09-06:
Seems related: https://github.com/espressif/arduino-esp32/issues/5048

ChuckMash, 2021-09-06:
Got it working!
I had to add a delay(1) to the main WLED::loop(), so far everything I've tested still works.
I'll tidy some stuff up and wrap this up.

blazoncek, 2021-09-06:
You should avoid `delay(1)` since it incurs 1ms delay which may hinder effect rendering.
If you want to reset WDT please use `yield()` instead.

GPIO pins 6-11 are SPI RAM pins which cannot be normally used for any other peripheral (well, under certain conditions they can). Check if by pin 9 you indeed mean GPIO9.

ChuckMash, 2021-09-06:
Seems like the delay(1) in this case is necessary as per the errors and link in my previous comments.
And ultimately, as an analog strip with an effective address count  of 1, most of the effects are already hindered to begin with!

Sorry, yes, GPIO9. Works to control the second white channel when the ESP is flashed with tasmota, so I know it can function at least at a hardware level.

blazoncek, 2021-09-06:
`yield()` does the same as `delay(0)` which does not wait 1ms.

As for GPIO9: I have experimented with WROVER module and PSRAM which also uses SPI but requires GPIO16 and GPIO17 and although most of the time using those for LED output worked they would cause ESP to reboot every so often.

ChuckMash, 2021-09-06:
I will see if yield will keep the watchdog reset boot loop at bay.

As for GPIO9, that's what I get for trusting dubious documentation. The documentation was incorrect.
In actuality, having just confirmed this with WLED, The white channels are controlled by GPIO2 and GPIO4.

blazoncek, 2021-09-06:
BTW, do you have a link for the device you bought?

ChuckMash, 2021-09-06:
No luck with adding another yield() to the loop, the delay(1) seems to be the only [solution](https://github.com/espressif/arduino-esp32/issues/5048#issuecomment-818043530) to keep the solo-1 running stable.

I'm not certain which precise version, as there's not much in terms of identification between older and newer models, but the controller and strip are the [WiZ LED Strip](https://www.wizconnected.com/en/consumer/products/8719514552586/)

I'm not certain the white temperature slider in WLED works correctly when using PWM RGBWC.
When sliding to cool white, the cool white LEDs are active, but when sliding to warm white, the cool white LEDs turn off, but then the warm white LEDs do not turn on.

ChuckMash, 2021-09-06:
Pins for reference.
```
Red - 25
Green - 26
Blue - 27
Cool White - 4
Warm White - 2
Pushbutton - 16
```

blazoncek, 2021-09-06:
WW CW pair is not yet fully supported in WLED.
The temperature slider only affects RGB(single)W channels.

ChuckMash, 2021-09-10:
I'm going to conclude this for the time being.

Support for the ESP32-SOLO-1 / WiZ Analog LED Strip is possible, but until this ESP32 variant is better supported in core, one must make some modifications to get it working.

[image]

[image]

In conclusion, the following steps are required to get WLED running on this hardware.

1.  Swap some libraries out: https://github.com/lbernstone/arduino-esp32-solo to support the single core ESP-SOLO-1
2. Add in a `delay(1)` at the end of the main loop: https://github.com/Aircoookie/WLED/blob/1d4487b6cd7c9a69bbce45c14ae1fb1c622e1d0e/wled00/wled.cpp#L121
Otherwise the ESP won't be able to multitask.
3. Attach to the ESP32-SOLO-1 RX/TX/GND pins for serial communication and Flashing.
4. Bridge IO0 to GND before booting to to being flashing process
5. Compile for ESP32 (with the aforementioned library swap)
6. Cross your fingers and Flash
7. Remove IO0 from Ground and power cycle
8. WLED should boot if successful.
9. Set LED output type to `PWM RGBWC` with pins 25, 26, 27, 4, 2
10. Set Button 0 pin to 16
11. Enjoy!

closed 2021-09-10 by ChuckMash

Awe7, 2021-11-23:
Can you upload firmware? I can't compile

Awe7, 2021-11-23:
> I'm going to conclude this for the time being.
>
> Support for the ESP32-SOLO-1 / WiZ Analog LED Strip is possible, but until this ESP32 variant is better supported in core, one must make some [...]

https://github.com/Aircoookie/WLED/issues/2171#issuecomment-917262843

> I'm going to conclude this for the time being.
>
> Support for the ESP32-SOLO-1 / WiZ Analog LED Strip is possible, but until this ESP32 variant is better supported in core, one must make some [...]

Can you upload firmware? I can't compile

Keyes, 2022-09-02:
Any news on esp32-solo-1 support? Sounds like it should be quite simple to integrate?

techninja, 2023-10-04:
Another year passes, and no replies. In the intervening years it seems getting this to build and keep the dependencies happy isn't simple (or at least it wasn't for me). @ChuckMash Thank you for your hard work, I've personally been able to get the hardware serial working, but compiling this with single core support seems broken and unsupported. You still in the field? Willing to try to compile this again?