wled/WLED #2173: How i Convert index.html as binary ? (i need to customize index page)

sanjulanipun, 2021-09-03:
**i need to know ho i can covert index page as binary?**
**like this ? i need to customize index page and after covert binary and after i need to upload normlly code. anyone know how to convert html page like this ?**

[image]

blazoncek, 2021-09-03:
https://kno.wled.ge/advanced/custom-features/

closed 2021-09-03 by blazoncek

sanjulanipun, 2021-09-03:
> https://kno.wled.ge/advanced/custom-features/

**ok, my problem is ok. but please i need to know how i can covert html code like this binary? this question for my knowlage. please tell me any tool or something how i can covert html code like this binary code. plz.**

blazoncek, 2021-09-03:
Read the article. It details exactly what to do.