wled/WLED #2180: Color changes gradually from first to last

SebAugh2004, 2021-09-07:
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

SebAugh2004, 2021-09-07:
I have 4 ws2812b light strips at 30led a meter 5 meters long and I am powering them with a 5 v 30 a power supply they are controlled by esp8266. When I chose the color white the first strip is white and the last strip flickers red and the esp8266 over heats

ChuckMash, 2021-09-08:
Sounds like it could be voltage drop.
Are you injecting power between each strip, or only at the beginning?

SebAugh2004, 2021-09-08:
In the middle

ChuckMash, 2021-09-08:
Something like [this?](https://peyanski.com/wp-content/uploads/2020/10/ws2812b-chain-my-way.jpg)

It might help if you shared a diagram of the wiring, or a picture of the problem.
It does sound to me like it could still be [voltage drop](https://forum.arduino.cc/t/led-strip-white-color-turns-into-yellow-at-the-end/314587)

blazoncek, 2021-09-08:
Please see Wiki or [WLED Docs](http://kno.wled.ge) for wiring and power injection instructions.

closed 2021-09-08 by blazoncek