wled/WLED #2181: wled json api on pyhton/micropyhton

nickpesman, 2021-09-09:
I am working on a little project which requires me to use the json api from wled. I am using a esp32 running micropython. And i have been stuck on this for a while now. When i send a get command to the api i get nothing in return. Or it just bugs and errors and i can't seem to figure out why. when i do get request to a json test website it is fine but i doesn't work on wledip/json for some reason. does someone have any idea how i van fix this? Or have something else i could try?

A simplified version of my code:
[image]

ChuckMash, 2021-09-10:
Can you show some less simplified code, or the error?

nickpesman, 2021-09-10:
Yea sure
[image]

Here is my code. This is what i get when i try to call the wled-ip/json url. It gives mee a 200 response code en then just stops

[image]

Here is te same code but here i call a json test website httpbin.org and it works just fine so i don't know what i am doing wrong here.

ChuckMash, 2021-09-10:
Does networkcred.ipaddresswled translate to something like http://4.3.2.1/json ?

Not sure about how micropython urequests lib differs from regular pythons requests lib.
In any case, not sure if this is related to WLED.

blazoncek, 2021-09-11:
Include `"v": true` in your JSON requests to get verbose response.

Normally GET-ting /json endpoint will return state and info objects, as well as palettes and effects.

nickpesman, 2021-09-12:
Do you mean something like this?
[image]

because this doesn't work for me....

blazoncek, 2021-09-12:
I said JSON. I have no clue how to create JSON object in Python. I do not know anything about Python for that matter.
And this may entirely be your particular Python problem, not WLED problem.

In any case use `curl` to test your setup and IPs used.

`$ curl http://wled.local/json` where _wled.local_ would be your WLED device IP address or mDNS name.

The response should look something like:
`{"state":{"on":true,"bri":128,"transition":10,...},"info":{"ver":"0.13.0-b2","vid":2109101,"leds":...}`

closed 2022-04-16 by Aircoookie