wled/WLED #2182: Macro for Power Off

GrillinGorilla, 2021-09-10:
Hi there, I'm looking to see if there's a macro command equivalent to powering off the LEDs as well as the blue light on the ESP. I can power it all off from the WLED app's Home Screen, but I can't schedule it from a macro. 

Thank you

blazoncek, 2021-09-11:
T=0
... or ...
{"on": false}

For all other info on API look into [WLED docs](https://kno.wled.ge).

closed 2021-09-11 by blazoncek

GrillinGorilla, 2021-09-12:
Hi there, 
I've been reading a lot of documentation and I think I asked my question incorrectly. 

I'm using version Toki (13.0 beta) and the Time & Macros settings does not include an Advanced Macro section. 

I am trying to power off the ESP from Time Controlled Presets section. So far, I can not achieve this. I can have a preset to set the strip to black and 0 brightness, but the small, blue LED on the ESP8266 chip remains on.  

[image]

blazoncek, 2021-09-12:
As I said: `T=0` ... or ... `{"on":false}`
Both of these **Presets** will be able to turn WLED off. And then assign that Preset to a time controlled presets table.

sansillusion, 2021-09-12:
Why is the development modules led such a problem ? Just put the esp in an
enclosure or cover the led with tape. All that forcing the blue led off has
done is created bugs in wled. Since they force it off when off LEDs flicker
etc... I preferred when signal was held low (keeping blue led on) that way
the strip did not have issues with flickering when off.

blazoncek, 2021-09-12:
If you use GPIO1 on-board LED is not lit.

GrillinGorilla, 2021-09-12:
> As I said: `T=0` ... or ... `{"on":false}`
> 
> Both of these **Presets** will be able to turn WLED off. And then assign that Preset to a time controlled presets table.

Sorry it took me awhile. After your last response, I learned there's an API command section in the body of the Preset. I got it all to work now. Thanks for your patience and guidance.

GrillinGorilla, 2021-09-12:
> Why is the development modules led such a problem ? Just put the esp in an
> enclosure or cover the led with tape. All that forcing the blue led off has
> done is created bugs in wled. Since they force [...]

Because my application requires a clear enclosure and I didn't want to use tape.