wled/WLED #2187: Version 210910 the infrared remote control cannot be used

neckyaan, 2021-09-11:
Version 210910, when the IR mode is JSON REMOTE, the infrared remote control cannot be used. my  infrared remote control is 44-keys,It works fine in 44-key RGB mode.

Aircoookie, 2021-09-11:
Hey! Thank you for the report. Can you share your `ir.json` file?

blazoncek, 2021-09-11:
Confirmed. Recent modifications trigger "win&0" command.

Aircoookie, 2021-09-11:
Realized what I have broken, thanks for confirming :) Fix incoming.

Aircoookie, 2021-09-11:
Fixed! I just saw the same value was assigned to two different variables, so I removed one of them, thinking they were redundant. I didn't catch that one of the vars was of type `const char*` while the other was `JsonObject.`
Please reopen if it still does not work with latest master.

closed 2021-09-11 by Aircoookie

neckyaan, 2021-09-12:
{
  "remote": "44-key",
  "0xFF3AC5": {
    "label": "Bright +",
    "pos": "1x1",
    "cmd": "!incBrightness"
  },
  "0xFFBA45": {
    "label": "Bright -",
    "pos": "1x2",
    "cmd": "!decBrightness"
  },
  "0xFF827D": {
    "label": "Next",
    "pos": "1x3",
    "cmd": "P1=1&P2=12&PL=~"
  },
  "0xFF02FD": {
    "label": "triger",
    "pos": "1x4",
    "rpt": true,
    "cmd": "T=0"
  },
  "0xFF1AE5": {
    "label": "Red",
    "pos": "2x1",
    "cmnt": "Lava",
    "cmd": "FP=8&CL=hFF0000&C2=hFF7F00&C3=hCC3D60"
  },
  "0xFF9A65": {
    "label": "Green",
    "pos": "2x2",
    "cmnt": "Forest",
    "cmd": "FP=10&CL=h8000&C2=h8040&C3=h30661E"
  },
  "0xFFA25D": {
    "label": "Blue",
    "pos": "2x3",
    "cmnt": "Breeze",
    "cmd": "FP=15&CL=hFF&C2=h7F00FF&C3=h3D60CC"
  },
  "0xFF22DD": {
    "label": "White",
    "pos": "2x4",
    "cmd": "FP=5&CL=hFFFFFF&C2=hFFFFFF&C3=hCCCCCC"
  },
  "0xFF2AD5": {
    "label": "Tomato",
    "pos": "3x1",
    "cmd": "FP=5&CL=hFF6347&C2=hFFBF47&C3=hCC646F"
  },
  "0xFFAA55": {
    "label": "LightGreen",
    "pos": "3x2",
    "cmnt": "Rivendale",
    "cmd": "FP=14&CL=h90EE90&C2=h90EEBF&C3=h96BE89"
  },
  "0xFF926D": {
    "label": "DeepBlue",
    "pos": "3x3",
    "cmnt": "Ocean",
    "cmd": "FP=9"
  },
  "0xFF12ED": {
    "label": "FloralWhite",
    "pos": "3x4",
    "cmnt": "Warm White",
    "cmd": "FP=5&CL=hFFFAF0&C2=hFCFFF0&C3=hCCC7C3"
  },
  "0xFF0AF5": {
    "label": "Orange",
    "pos": "4x1",
    "cmnt": "Sakura",
    "cmd": "FP=49&CL=hFFA500&C2=hD9FF00&C3=hCC753D"
  },
  "0xFF8A75": {
    "label": "Turquoise",
    "pos": "4x2",
    "cmnt": "Beech",
    "cmd": "FP=22&CL=h40E0D0&C2=h40A0E0&C3=h59B393"
  },
  "0xFFB24D": {
    "label": "Purple",
    "pos": "4x3",
    "cmd": "FP=5&CL=h800080&C2=h800040&C3=h541E66"
  },
  "0xFF32CD": {
    "label": "LavenderBlush",
    "pos": "4x4",
    "cmd": "FP=5&CL=hFFF0F5&C2=hFFF2F0&C3=hCCC3C8"
  },
  "0xFF38C7": {
    "label": "Yellowish",
    "pos": "5x1",
    "cmnt": "Orangery",
    "cmd": "FP=47"
  },
  "0xFFB847": {
    "label": "Cyan",
    "pos": "5x2",
    "cmnt": "Beech",
    "cmd": "FP=22&CL=hFFFF&C2=h7FFF&C3=h3DCCA8"
  },
  "0xFF7887": {
    "label": "Magenta",
    "pos": "5x3",
    "cmd": "FP=5&CL=hFF00FF&C2=hFF007F&C3=hA83DCC"
  },
  "0xFFF807": {
    "label": "GhostWhite",
    "pos": "5x4",
    "cmd": "FP=5&CL=hF8F8FF&C2=hFBF8FF&C3=hC8C9CC"
  },
  "0xFF18E7": {
    "label": "Yellow",
    "pos": "6x1",
    "cmd": "FP=5&CL=hFFFF00&C2=hFFC800&C3=hFDFFDE"
  },
  "0xFF9867": {
    "label": "Aqua",
    "pos": "6x2",
    "cmd": "FP=5&CL=hFFFF&C2=h7FFF&C3=h3DCCA8"
  },
  "0xFF58A7": {
    "label": "Pink",
    "pos": "6x3",
    "cmd": "FP=5&CL=hFFC0CB&C2=hFFD4C0&C3=hCCA8B7"
  },
  "0xFFD827": {
    "label": "LightCyan",
    "pos": "6x4",
    "cmd": "FP=5&CL=hE0FFFF&C2=hE0EFFF&C3=hBACCC7"
  },
  "0xFF28D7": {
    "label": "Red +",
    "pos": "7x1",
    "cmd": "FP=5&R=~16"
  },
  "0xFFA857": {
    "label": "Green +",
    "pos": "7x2",
    "cmd": "FP=5&G=~16"
  },
  "0xFF6897": {
    "label": "Blue +",
    "pos": "7x3",
    "cmd": "FP=5&B=~16"
  },
  "0xFFE817": {
    "label": "Quick",
    "pos": "7x4",
    "cmnt": "Fx speed +16",
    "cmd": "SX=~16"
  },
  "0xFF08F7": {
    "label": "Red -",
    "pos": "8x1",
    "cmd": "FP=5&R=~-16"
  },
  "0xFF8877": {
    "label": "Green -",
    "pos": "8x2",
    "cmd": "FP=5&G=~-16"
  },
  "0xFF48B7": {
    "label": "Blue -",
    "pos": "8x3",
    "cmd": "FP=5&B=~-16"
  },
  "0xFFC837": {
    "label": "Slow",
    "pos": "8x4",
    "cmnt": "FX speed -16",
    "cmd": "SX=~-16"
  },
  "0xFF30CF": {
    "label": "Diy1",
    "pos": "9x1",
    "cmd": "PL=1"
  },
  "0xFFB04F": {
    "label": "Diy2",
    "pos": "9x2",
    "cmd": "PL=2"
  },
  "0xFF708F": {
    "label": "Diy3",
    "pos": "9x3",
    "cmd": "PL=3"
  },
  "0xFFF00F": {
    "label": "Auto",
    "pos": "9x4",
    "cmnt": "Toggle preset cycle",
    "cmd": "CY=2&PT=60000"
  },
  "0xFF10EF": {
    "label": "Diy4",
    "pos": "10x1",
    "cmd": "PL=4"
  },
  "0xFF906F": {
    "label": "Diy5",
    "pos": "10x2",
    "cmd": "PL=5"
  },
  "0xFF50AF": {
    "label": "Diy6",
    "pos": "10x3",
    "cmd": "PL=6"
  },
  "0xFFD02F": {
    "label": "Flash",
    "pos": "10x4",
    "cmnt": "Cycle Effects",
    "cmd": "FX=~"
  },
  "0xFF20DF": {
    "label": "Jump3",
    "pos": "11x1",
    "cmnt": "Colortwinkles",
    "cmd": "PL=7"
  },
  "0xFFA05F": {
    "label": "Jump7",
    "pos": "11x2",
    "cmnt": "Sinelon Dual",
    "cmd": "PL=8"
  },
  "0xFF609F": {
    "label": "Fade3",
    "pos": "11x3",
    "cmnt": "Rain",
    "cmd": "PL=9"
  },
  "0xFFE01F": {
    "label": "Fade7",
    "pos": "11x4",
    "cmnt": "Lighthouse",
    "cmd": "FP=~"
  }
}

commit bd239428 (2021-09-11): Fixed IR JSON cmd string (closes #2187 )