wled/WLED #2188: Can-Bus integration

Aircoookie, 2021-09-11:
"it would be great if one could change the presets via canbus because the ESP32 has integrated canbus"

By @badFreddy0815 in #2186

adamryczkowski, 2024-11-16:
Hi!

We are about to add a Can-Bus integration for ESP32, connected through TJA1050 CAN transceiver. We do not have large requirements - just to use the Can Bus to change the presets.

We are not sure how exactly to integrate it to WLED - Our fist idea is to integrate it through adding a glorified custom button, since they are good for preset changes. 

Do you have any other thoughts about how to do it? 

I know that there is such a thing as usermods, but I can't find where the class interface (class `Usermod`) is documented. 

In particular, I can't find information if it is possible for the usermod to define events (a Can Bus packet has arrived/button was pressed).

adamryczkowski, 2024-11-17:
Update:

I found a `usermod_v2_example.h` that serves as documentation. That sealed the decision to go through by writing a usermod.

netmindz, 2024-11-26:
I would advise extreme caution with any integration that tries to write data to CanBus, you won't want WLED to interfere with the critical systems in the car. Listening and responding to the rev counter should be ok, but still be sure you know exactly how your integration might impact other systems

adamryczkowski, 2024-11-26:
:-D

@netmindz You've brightened my day :-D I never wrote I am going anywhere near any car with my WLED installation. I am definitely never going near to interfere with my car :-).  We are building a CAN-BUS network at home, _dedicated_ for WLED and a few temperature sensors and similar stuff. 

We simply hate unreliability and difficult to maintain security of WiFi connections, and prefer a wired connection.

Thank you for your concern, though. It is a valid point for anyone else reading this post.

BTW, we have a working prototype already!