wled/WLED #2190: Upload output from xlights will not change pin number for the first pin in wled.

processor83, 2021-09-12:
**Describe the bug**
When i use upload output in xlights it will push pin configuration but will not change the pin number for the first output in wled.

**To Reproduce**
In xlights put a promp on port 1 in the visdualiser and then close visualiser and push upload output.
In wled if you look at the led config page you will see that first pin in wled is not changed accoprding to what is sent by xlights.  other ports are working.

**Expected behavior**
That all output accept the config sent by xlights.

**WLED version**
 - Board:ESP32
 - Version 0.13.0-b2
 - Format bin from release

**Additional context**
Maybe allow a way to also ignor those pin number thata are pushed since not everybody use same pins but now xlights have pre defined pins for generic wled and quinled. 

This is also something that must be seen adressed on xlights side
Thank you for your help!

blazoncek, 2021-09-12:
What does Xlights send to WLED?
If the format is correct there are no obstacles to change required pin.

processor83, 2021-09-12:
I will check how I can see what xlights send. It's true that it also can be on the xlights side.

I'm sorry i don't know a lot about programming will try to use wire shark to see what it send.

processor83, 2021-09-12:
> What does Xlights send to WLED?
> 
> If the format is correct there are no obstacles to change required pin.

I did some test and I must be right xlights read current pin config and re use it so that why it will not change first output since it always have one to read.  Also it mean that ip pins are setup it will not over write them!

blazoncek, 2021-09-12:
I'm not sure I understand correctly, but if I do, you should talk to Xlights team.

processor83, 2021-09-12:
You can close this. It's true it's the way xlights do the config.

Xlights read pin config and if there is already a pin configured it will re use it.  Since the first led output cannot be blank in wled it will never change this pin the way they do it but it is also a good thing.  Sorry I did not understood how it was done now that I looked at wire shark capture it's all clear.

closed 2021-09-13 by blazoncek