wled/WLED #2192: Timezone info now incorrect from NTP

GSVNoFixedabode, 2021-09-12:
I noticed on the weekend my LED set was turning off an hour early, then realised it was also turning on an hour early.  Checked the other one (60-LED ring) and realised that was an hour fast as well.  

Both are set to 'NZST/NZDT' time and using 0.WLED.NTP server and were working fine up until the weekend.

My guess is that the WLED.NTP server has incorrect Daylight Savings dates loaded.

blazoncek, 2021-09-12:
I noticed that too on couple of occasions but goes away after a few days.

pbolduc, 2021-09-17:
I believe NTP servers always serve up UTC dates. Each device is required to convert to their correct UTC offset.

GSVNoFixedabode, 2021-09-28:
Daylight Saving rolled around here on the weekend (now on NZDT) and WLED now showing correct time.  Not sure if it's the WLED NTP server or baked into WLED itself but the changeover dates for NZ daylight savings are incorrect.

pbolduc, 2021-09-28:
I always prefer the zone specific addresses from https://www.ntppool.org/en/   In New Zealand, there are 29 active IPv4 servers https://www.ntppool.org/zone/nz

1) check your npt server settings
2) check your time zone settings

The code is supposed to check ntp twice a day.

GSVNoFixedabode, 2021-09-28:
Switched to NZ pool. thanks

closed 2021-09-28 by GSVNoFixedabode