wled/WLED #2193: Add all possible timezones

cmtmc, 2021-09-13:
**Is your feature request related to a problem? Please describe.**
Timezones are not for all possible places and don't have number values. Help says "Your time zone. Open an issue if yours is unsupported."

**Describe the solution you'd like**
Make it like Windows way - numbers and geolocations - from UTC-12:00 International Date Line West till UTC+14:00 Kiritimati island. Including utc-9:30, utc+9:30, utc+12:45.

blazoncek, 2021-09-13:
Please provide a complete list and I'll do my best.

cmtmc, 2021-09-14:
Official site - https://docs.microsoft.com/en-us/windows-hardware/manufacture/desktop/default-time-zones

Sorted list - https://howtomanagedevices.com/windows-10/1774/list-of-windows-10-time-zone-codes-tzutil/

blazoncek, 2021-09-14:
Both of these lists lack information on exact DST change.

cmtmc, 2021-09-14:
https://en.wikipedia.org/wiki/Daylight_saving_time_by_country
https://www.timeanddate.com/time/dst/2020.html - dates also differ

DST depends by country, not by timezone. Say Russia has 11 timezones - and "Observed DST in 1917–1919, 1921 (some areas), and 1981–2010. In 2011–2014, used permanent DST. In 2014, left permanent DST and switched to permanent standard time."

Not sure whether supporting DST is needed in WLED on regular basis.

Aircoookie, 2021-09-14:
DST is the sole reason for timezones IMO, for regions with a year round standard time, googling `[country] utc offset` and putting that into the UTC offset field in the settings is a lot easier than even selecting the correct timezone from a dropdown menu.

Other than that, I would love to have support for every timezone. Windows `tzutil /l` lists 139, many of which are duplicates or "rare" time zones. While the time zone system in WLED was reworked to hardly use any RAM, of course we need flash, about 36 bytes per timezone for the dropdown entry and about 8 bytes for the timezone definition. That would mean around 6kB extra flash usage, something to consider. The effect of that would be lessened as soon as settings pages are gzip compressed (TODO). 

I would support adding every _distinct_ time zone, that is, every combination of UTC offsets and DST that are in actual use today.
E.g. Japanese and Korean standard time is the same (UTC+9 year round), so I would regard them as one time zone.

blazoncek, 2021-09-14:
If you want to have proper _time controlled presets_ DST is essential.
Otherwise we could simply use UTC +/- offset notation.

pbolduc, 2021-09-17:
It is user friendly to support time zone names cause that is what people know. Never going to solve everyone's asks. Maybe 85%? Having extended UI (hidden by default) allow specify custom time zone, Standard Time Offset, Daylight Offset, and the parameters (1st/2nd/3rd, day of week, in month) would allow them to set any time zone settings.  Better yet, would be nice if could pull the time zone from the internet (assume has internet access) for setting the parameters (browser CORS request?) so the settings are pulled and updated in UI and then posted to device?  Changing time zones is something we will continue to chase, - https://data.iana.org/time-zones/tzdb/NEWS

cmtmc, 2021-09-20:
> It is user friendly to support time zone names cause that is what people know. 

If the person finds WLED, chooses power supply, purchases components, solders diode/capacitors/step-up controller - I believe he can find out numeric value of his timezone..

Natal-CWB, 2021-09-28:
It would be nice to have all timezones but it will consume a lot of precious memomry from ESP. Why not set UTC in hours? just add a 3600 multiplier to the code, because it's a little weird to put it in seconds. Just use UTC -3, UTC +13, UTC -6,5 instead of UTC-10800, UTC +46800, UTC -23400

blazoncek, 2021-09-28:
Good point.

pbolduc, 2021-09-28:
The reason the time is set in seconds is because it is following unix timestamp format - https://www.unixtimestamp.com/   "This count starts at the Unix Epoch on January 1st, 1970 at UTC"  Logically, I think WLED is only looking at the current time or short times in the future.  There is no reason WLED needs to use the Unix Epoch.  That could be shifted to WLED Epoch of Jan 1st, 2020. This only represents when "time began".

As for the problem of the time zone rules.  The code uses this struct to capture the rules, requires 1 or 2 of these per time zone.

```c++
struct TimeChangeRule
{
    uint8_t week;      //First, Second, Third, Fourth, or Last week of the month
    uint8_t dow;       //day of week, 1=Sun, 2=Mon, ... 7=Sat
    uint8_t month;     //1=Jan, 2=Feb, ... 12=Dec
    uint8_t hour;      //0-23
    int16_t offset;        //offset from UTC in minutes
};
```

Based on the current time zones (https://en.wikipedia.org/wiki/List_of_UTC_time_offsets) the offset needs to go from -12 to +14 hours.  Not all time zones are full hours, some are 1/2 hour or 15/45 minutes. So storing the offset in minutes makes sense.  If using bit packing and offset measured in multiples of 15 minutes, the size per time zone rule could go from 6 bytes down to 4 bytes

```
week:         0 -   4  =          3 bits
dow :         1 -   7  =          3 bits
mon :         1 -  12  =          4 bits
hr  :         0 -  23  =          5 bits
offset:    -720 - 840  =
     (15m)  -48 -  56  =         12 bits
                               ---------
                                 27 bits
                                  4 bytes
```

Based on my Googling, there are 38 time zones - https://www.timeanddate.com/time/current-number-time-zones.html

38 * 6 * 2 = 456 bytes for rules, not to mention the required markup in the HTML front end, that would be reduced to 304 bytes if using the bit packing. The UI stuff *may* be able be able to be packed if used javascript to render the list instead of hard coded html markup.

softhack007, 2023-10-07:
@blazoncek is there anything left to do for this?
If yes, we might be able to complement our TZ list in 0.14.1.

blazoncek, 2023-10-07:
I given up on the idea of implementing this. Do what you think is ok.