wled/WLED #2194: mDNS is not working on 0.13.0

LowderPlay, 2021-09-14:
**Describe the bug**
mDNS service discovery is broken on 0.13.0 betas. The device can't be reached through its hostname in the browser, resolved using `avahi-browse` or `avahi-resolve` or discovered through the app. 

**To Reproduce**
Install 0.13.0 beta. Try to discover the device in the app, or visit its webpage in the browser.

**Expected behavior**
The device can be successfully resolved in 0.12.0.

**WLED version**
 - Board: ESP8266 (ESP-12F module)
 - Version 0.13.0
 - Format: Binary

blazoncek, 2021-09-14:
Hmm. This is working on 26 of my ESPs.

LowderPlay, 2021-09-15:
I reverted back to 0.12.0 and the discovery worked flawlessly, but when I updated one more time, discovery stopped working for me.

blazoncek, 2021-09-15:
What environment do you run discovery on?

LowderPlay, 2021-09-15:
I tried using the app and `avahi-browse -alr` on a Linux device.

blazoncek, 2021-09-15:
There is a known bug in avahi and ESP implementation.
Can you check Win or mac?

LowderPlay, 2021-09-15:
I'm not sure how to try using it on windows. Could you suggest software that I can use? I tried using java library jmdns on windows, it doesn't work for me on the new version.

blazoncek, 2021-09-15:
Try BonjourBrowser for Windows or similar tool zeroconfServiceBrowser.

LowderPlay, 2021-09-15:
Huh. Weird, I can discover the lights now.

blazoncek, 2021-09-15:
Not weird at all. 😄

blazoncek, 2021-09-29:
Closing as it seems duplicate of #2009

closed 2021-09-29 by blazoncek