wled/WLED #2197: Consider using Framebuffer::GFX for 2D support

marcmerlin, 2021-09-16:
I looked at WLED in the past, but didn't use it because I use 2D and it was 1D only.
I see that you are adding 2D support now (cool).

Could you have a look at this?
* http://marc.merlins.org/perso/arduino/post_2020-03-16_Framebuffer_GFX_-Choosing-between-its-3-2D-APIs_-FastLED-XY_-NeoMatrix_-and-LEDMatrix_-and-detail-of-its-many-supported-hardware-backends.html
* https://github.com/marcmerlin/Framebuffer_GFX

This will give you 2D support on multiple hardware backends (TFT, RGBPanel, FastLED, etc...)
It will also give you 3 APIs, FastLED, LEDMatrix, and Adafruit::GFX

blazoncek, 2021-09-18:
How do you think this will fit to WLED?
WLED doesn't use FastLED library for displaying pixels.

marcmerlin, 2021-09-18:
Framebuffer::GFX only uses FastLED to get the CRGB class definition, which is used to store the framebuffer, but you're right that is an extra dependency if this is an issue for you.
I currently use it with RGBPanels with 2 different backends, and TFTs, and the rest of FastLED isn't used or imported.
That said, if that is a concern to you, it should be possible to import in a couple of FastLED include files so as to remove the FastLED dependency.
That said, it does add a dependency on Adafruit::GFX, which of course is by design since it adds support for its 2D functions.

blazoncek, 2021-09-18:
Thanks for clarifying. WLED uses FastLED in a similar way for some calculations, but doesn't use CRGB buffers.
Actually all bufers are handled by NeoPixelBus.

marcmerlin, 2021-09-18:
Right, so I'll leave this up to you, but basically instead of doing it the hard way and re-implementing a bunch of 2D stuff again, integrating with existing APIs (Adafruit::GFX and LEDMatrix) takes all that non fun and boring work away. 
You could take my lib and modify it to be based on top of NeoPixelBus's version of what I'm hopeful is also an array of RGB888 (which is what FastLED's CRGB really is).
Another thing that will probably be even simpler if neopixelbus pixel array is indeed compatible with FastLED (i.e. RGB888), then you could really use my code as is with the FastLED definitions and cast the pointer to the neopixelbus array. 
That way you get both APIs against the same LED storage, and you gain the extra FastLED class methods like dim and so forth (but more importantly you also get compatibility with Adafruit::NeoMatrix which gives you all of Adafruit::GFX)

blazoncek, 2021-09-18:
Unfortunately there is no array exposed fromNeoPixelBus. You manipulate pixels by function calls.

marcmerlin, 2021-09-18:
Ok, that's a bit unfortunate.
If I had to work on this, I'd modify NPB to add a method which returns a pointer to the array, you could then define 
CRGB *leds = (CRGB *) NPB.returnarray()
and everything else I described, should work.

embedded-creations, 2021-09-20:
> Unfortunately there is no array exposed fromNeoPixelBus. You manipulate pixels by function calls.

NeoPixelBus doesn't store CRGB data anywhere, it only stores the raw pixel data that gets sent out to the LEDs.  If using NeoPixelBrightnessBus (like WLED does), the raw pixel data is lossy, so `{254,254,254}` with brightness 50% would be read back as `{255,255,255}` if you read the raw pixel data back and reversed the brightness conversion.

Andrew Tuline's fork already has rectangular matrix 2D support, and adds a CRGB buffer so FastLED effects can be ported to WLED with less effort and without the slow and lossy conversion using NeoPixelBus:: getPixelColor(). 

https://github.com/atuline/WLED/blob/master/wled00/FX.cpp#L4330

The CRGB buffer is fed into NeoPixelBus at the end of the effects.

https://github.com/atuline/WLED/blob/master/wled00/FX.cpp#L4347

I haven't been keeping up with the latest 2D changes - they had an overhaul with a lot of improvements within the last few months.  I believe that you can now set the geometry/tiling of matrices using API commands instead of recompiling.  I'm pretty sure what they have now isn't Adafruit_GFX compatible, so that would be an improvement.

https://github.com/atuline/WLED/wiki/2D-Support

I thought the same thing (and came up with a convoluted way to get access to the raw pixels without modifying NeoPixelBus), but turns out there is a public method to expose the array:

https://github.com/Makuna/NeoPixelBus/blob/master/src/NeoPixelBus.h#L223-L226

The data is stored

marcmerlin, 2021-09-20:
Thanks Louis, this sounds like a lot more work than I thought it would/should be.
Dumb question: at some point, isn't easier to substitute FastLED in WLED and then you get the CRGB array and 2D + all the 2D APIs via framebuffer::GFX for free?

blazoncek, 2021-09-20:
Thi would require rbuilding core from scratch.
Unfeasible IMO.

embedded-creations, 2021-09-20:
NeoPixelBus has RGBW support, and you can choose pins etc at runtime, which I don't believe you can do with FastLED.  For WLED's requirements, I think NeoPixelBus is a better library for driving pixels than FastLED

marcmerlin, 2021-09-20:
understood. I'll then let you decide what you think is the least amount of work/pain.  You can indeed re-add 2D support from scratch but realistically by then you will have a full 2D framebuffer and have rewritten adafruit::GFX, or at least parts thereof.
Another idea: if there is no array of CRGB currently, would it be bad to use Framebuffer::GFX for its 2D framebuffer and all the APIs to write into it, and then implement the show() method to push the pixels into neopixelbus?  Afterall that's exactly what that class is designed to do: it holds the data and then you provide your own show() method that pushes the pixels to yoru backend, be it @embedded-creations's SmartMatrix, or FastLED, or a TFT, or even rpi-rgb-panel on rPi (which is what I'm currently using)

marcmerlin, 2021-09-20:
one implementation of a superclass for a TFT on top of framebuffer::gfx
https://github.com/marcmerlin/FastLED_ArduinoGFX_TFT/blob/f055f0d3d1ee9ae6121ebdc1209c701a4d565530/FastLED_ArduinoGFX_TFT.cpp#L57

embedded-creations, 2021-09-20:
I'm not working on rectangular matrix 2D support right now, and I think the team working on the atuline fork already has a solution that works for them (and they already have an array of CRGB, I linked to above).  If you think you have something they're missing and they don't respond to the GitHub Issue you made, you could join their Discord (linked in their README or wiki) where they're most active and make a recommendation.

marcmerlin, 2021-09-20:
Thanks Louis. That's probably more than the time I have available, especially as I don't use WLED (and that was mostly indeed because it didn't support 2D or fastled API, which I rely on) and settled on https://github.com/marcmerlin/OmEspHelpers for bare wifi/basic webpage support.

You probably know some about the pros and cons, so I'll leave it up to you or others with a vested interest, to make the decision that makes the most sense for the project.

ewoudwijma, 2021-09-21:
Hi there, I did most of the changes in WLED Soundreactive, building on top of the 0.12 version, ready for the 0.13 version, to be released in the near future. The current dev version contains the latest: https://github.com/atuline/WLED/tree/dev. There is pretty much covered right now including sound reactive 2D effects. You can install this on an esp32 attach a matrix to it and have a comparison with your code. I need to find the time to look at your code but I am curious what you have. I agree with above commenters that a lot is in place already and we should not change the core fundamentals .

atuline, 2021-09-21:
I"m definitely deferring to Ewoud and embedded-creations as this expanded 2D work has grown beyond my pay grade. Now that we have the little FastLED wrapper, life is much easier for me with any FastLED animations I write. Furthermore, RGBW support is a nice bonus. I have, however, been following (and been inspired by) Marc's various postings ever since the G+ pages of old.

blazoncek, 2021-09-29:
Closing due to current infeasibility.
Feel free to reopen if core changes or there is any advancement making it easier to port to WLED.

closed 2021-09-29 by blazoncek