wled/WLED #2198: WeMos D1 R2 ESP8266 + 32 Mb flash WiFi RobotDyn + WLED ESP8266 did not work

Inputselector, 2021-09-16:
I bought WeMos D1 R2 ESP8266 + 32 Mb flash WiFi RobotDyn with the idea of uploading WLED. Unfortunately, nothing works. I am uploading the WLED_0.13.0-b2_ESP8266.bin firmware. The installation process itself ends correctly. Unfortunately, nothing can be done anymore because there is no wifi network. The wifi module itself is functional because before uploading the WLED, the default name of the device's wifi network appeared. Will it work at all or should I buy other controllers?

Inputselector, 2021-09-18:
WiFi was activated, the WLED-AP access point appeared after flashing and restarting the device. The only problem is that I was able to do this on the software version [WLED version 0.5.0] WLED_0.5.0_ESP8266_RGB_1802222.bin. WiFi does not turn on on a newer version of the software.

blazoncek, 2021-09-18:
Have you tried erasing flash prior to uploading firmware?

Inputselector, 2021-09-18:
I erase flash memory each time I upload new firmware. The case is that probably the oldest available version of WLED [WLED version 0.5.0] has something that has been changed in the next version and each newer version, so it does not want to work on my device. The access point started working, it was possible to ping it and it responded, but the configuration window cannot be opened, the application also does not find it. Everything on the 0.5 WLED version.

Aircoookie, 2022-04-16:
@Inputselector if you are still interested, try this binary: https://drive.google.com/file/d/1joU3ZmxjLTRr_4ZtWEjyZ4GFtK8186BO/view?usp=sharing
Note that OTA updates will not work with this specific build.