wled/WLED #2204: Shelly2 Firmware doesn't exist anymore

SaPoX76, 2021-09-18:
The Wiki-->Compatible hardware say i have to use an forked Firmware:
* https://github.com/srg74/WLED/tree/ShellyRGBW2
* https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/Shelly_RGBW2

Both links are dead :(

Which Hardware should i use now for a Shelly RGBW2?

SaPoX76, 2021-09-18:
Which <s>Hardware</s> Firmware should i use now for a Shelly RGBW2?

blazoncek, 2021-09-18:
You can kindly ask @srg74 to provide them again.

al3x1337, 2021-09-20:
He commented in Discord a couple days ago that the esp01 (esp8266 version i assume) bin file works with only the pins needing to be configured after words. Have not tried this out myself yet.

SaPoX76, 2021-09-20:
Thanks for your answers!

Yesterday i deceided to use the Shelly for an other Projekt, with an "old-school" LED-Stripe.
Today i have used a NodeMCU and it works fine.

Should I close this thread, or wait for more replies, for someone facing the same problem?

al3x1337, 2021-09-21:
> 
> 
> Thanks for your answers!
> 
> Yesterday i deceided to use the Shelly for an other Projekt, with an "old-school" LED-Stripe.
> Today i have used a NodeMCU and it works fine.
> 
> Should I close this thread, or wait for more replies, for someone facing the same problem?

You can close the thread after this response as I have flashed WLED on a Shelly RGBW2 today and will outline my steps below with serial flashing:

1) Attached striped Ethernet cable into the serial pins on the RGBW2 and attach those wires to a USB-to-UART adapter. Use this as a guide for the pins: https://tasmota.github.io/docs/devices/Shelly-RGBW2/#serial-connection. Remember to switch TX and RX from USB to RGBW2 and ground GPI0 when plugging in the USB flasher.

2) Download the WLED_0.12.0_ESP8266_1M.bin file and flash using Pyflasher
_I've had Error 10: could not mount filesystem!" "SyntaxError: UnexpectedToken in JSON at position <0" messages in the UI and Wifi settings not saving when just using the ESP8266 bin file. I believe the RGBW2 has 2MB RAM so you can use that file also but I have not tested. The main take away is that the default binaries for WLED are compatible with the RGBW2 without any modifications._  

3) Once WLED is flashed go to LED preferences -> LED outputs -> select PWM RGBW -> color order: RGB -> pins 12, 15, 14, 4 -> save.

That's all required for me to get WLED working.

SaPoX76, 2021-09-22:
Well done ... closed!

closed 2021-09-22 by SaPoX76