wled/WLED #2205: RGB  selection for Esp32

9ghost9, 2021-09-18:
The output is in reversed  order 0 is full on  and 255 does not fully turn off the leds .

@ 255 the leds flicker like mad and remain near 20 % brightness

**WLED version**
 - Board: [e.g.  ESP32 dev]
 - Version [WLED_0.13.0-b0_ESP32.bin]
 - Format [e.g. Binary, self-compiled]

Thank you for your help!

blazoncek, 2021-09-18:
Update to more recent build or compile yourself if possible.
Get a quality PSU with enough power for all LEDs and make sure you inject power to LEDs appropriately.

9ghost9, 2021-09-18:
> Update to more recent build or compile yourself if possible.
> Get a quality PSU with enough power for all LEDs and make sure you inject power to LEDs appropriately.

This is a problem with WLED code . I have written my own PWM code for rgb on the Esp8266 and the ESP32 and it works flawlessly. I am also driving AC leds, filament bulbs and 200W floods  with this same PWM  and an optocoupler .
AnalogWrite for ESp8266 and LedcWrite for ESP32  is the best- smoothest PWM dimming for common anode rgb .

I would like to add my code as a UserMod if this is not correctable by the dev team. I am just not certain how 
add and include using gpio .

blazoncek, 2021-09-18:
I am not sure if we talk about the same thing or that I understand what you want to say.

WLED supports analog LEDs in the following variants: PWM White, PWM WW/CW, PWM RGB, PWM RGBW and PWM RGBWW.
All of the variants work on ESP8266 and ESP32 using MOSFET (or appropriate bipolar transistors).

9ghost9, 2021-09-18:
> I am not sure if we talk about the same thing or that I understand what you want to say.
> 
> WLED supports analog LEDs in the following variants: PWM White, PWM WW/CW, PWM RGB, PWM RGBW and PWM RGBWW.
> All of the variants work on ESP8266 and ESP32 using MOSFET (or appropriate bipolar transistors).

Yes we are talking analog rgb . The WLED setting do not provide for inverted output . 0 setting is full bright on the esp32 that I have tested , and 255 is not off .
My testing was with WLED_0.13.0-b0_ESP32.bin. and adding a additional output to enable rgb selection .

I did not test compiling in the IDE .  I guess I should have asked if the .bin method of adding outputs for rgb is less refined compared to compiling a sketch in the IDE.

I seem to recall ver 09 did not have issues with rgb but that was compiled only at the time .

blazoncek, 2021-09-18:
Actually WLED supports inverted output for analog but it looks like this setting is not exposed in UI.
Will correct it.

9ghost9, 2021-09-18:
> Actually WLED supports inverted output for analog but it looks like this setting is not exposed in UI.
> Will correct it.
Thank you !
I will try ver 09 and compile in IDE and see if dimming AC works better .
I am using IGBT and mapping pixel data which works perfectly for Dimming AC fixtures/elements.
Simple PWM via Web Server works perfect also ,but the rgb in WLED does not give smooth dimming  and will not dim to off .

9ghost9, 2021-09-18:
BTW WLED is fantastic ! Hoping to use it as all in one control for both AC and DC elements year round.

9ghost9, 2021-09-20:
Compiled WLED ver 0.9.1 RGB dimming is perfect .

blazoncek, 2021-09-20:
For me, 0.13 works flawless. Dimming analog leds as expected.
You will need to wait until _Inverted_ option is inluded in master, though.

9ghost9, 2021-09-20:
The trouble is see on ESP32 Dev is limited range on brightness slider and there is a lot of flicker . This does not happen with ver ,0.9.1 ,it has very smooth dimming and full slider range 0 - 255 .
I will test ver 0.13 on Wemos D1 Esp32 and on Wemos D1 Esp8266 next to compare .

Thanks again for all your work making this a great firmware .

9ghost9, 2021-09-20:
Wemos D1 esp8266 no analog rgb dimming with  ver 0.13.0-b2.

Ver 0.12  works perfectly .

next test Wemos D1 ESP32  :)

blazoncek, 2021-09-21:
I can only say that you must be doing something wrong. :)

9ghost9, 2021-09-21:
> I can only say that you must be doing something wrong. :)

Hey my friend ,  if we are not doing things wrong , we are not learning of gaining experience :)

From what i have seen and experienced through my testings ver 0.13.x, the R,G,B analog selection setting  is not isolating the discrete r ,g and b channels. This could be due to non inversion of the data and the fact that the index defaults to 30 for some odd reason . However I am conceding to 0.13.x until the next update .

I am extremely impressed with  this version working as it does .
I have full range dimming of all types of AC fixtures  with the analog channels , just incredible !
Wled ver 0.12.0 does this isolation perfectly .
ver 0.91 does this perfectly also , but not as perfect as 0.12.0
.ver 0.13 .x  is no inversion ->no comment   .

Thanks again for the great firmware .

9ghost9, 2021-09-24:
Thank you for the latest release . Unfortunately I see the same issue with the Brightness Range Slider  and  Analog rgb

On the ESP32.

Does this still exist ?  If so , where can I find this to edit in the source code ?

`#define WLED_USE_ANALOG_LEDS //Uncomment for using "dumb" PWM controlled LEDs (see pins below, default R: gpio5, G: 12, B: 15, W: 13)
//#define WLED_USE_H801 //H801 controller. Please uncomment #define WLED_USE_ANALOG_LEDS as well
//#define WLED_USE_5CH_LEDS  //5 Channel H801 for cold and warm white

#define BTNPIN  0  //button pin. Needs to have pullup (gpio0 recommended)
#define IR_PIN  -1  //infrared pin (-1 to disable)  MagicHome: 4, H801 Wifi: 0
#define RLYPIN 12  //pin for relay, will be set HIGH if LEDs are on (-1 to disable). Also usable for standby leds, triggers,...
#define AUXPIN -1  //debug auxiliary output pin (-1 to disable)

#define RLYMDE 1  //mode for relay, 0: LOW if LEDs are on 1: HIGH if LEDs are on

//END CONFIGURATION

#if defined(USE_APA102) || defined(USE_WS2801) || defined(USE_LPD8806)
 #define CLKPIN 0
 #define DATAPIN 2
 #if BTNPIN == CLKPIN || BTNPIN == DATAPIN
  #undef BTNPIN   // Deactivate button pin if it conflicts with one of the APA102 pins.
 #endif
#endif

#ifdef WLED_USE_ANALOG_LEDS
  //PWM pins - PINs 15,13,12,14 (W2 = 04)are used with H801 Wifi LED Controller
  #ifdef WLED_USE_H801
    #define RPIN 15   //R pin for analog LED strip   
    #define GPIN 13   //G pin for analog LED strip
    #define BPIN 12   //B pin for analog LED strip
    #define WPIN 14   //W pin for analog LED strip 
    #define W2PIN 04  //W2 pin for analog LED strip
    #undef BTNPIN
    #undef IR_PIN
    #define IR_PIN  0 //infrared pin (-1 to disable)  MagicHome: 4, H801 Wifi: 0
  #else
  //PWM pins - PINs 5,12,13,15 are used with Magic Home LED Controller
    #define RPIN 5   //R pin for analog LED strip   
    #define GPIN 12   //G pin for analog LED strip
    #define BPIN 15   //B pin for analog LED strip
    #define WPIN 13   //W pin for analog LED strip 
  #endif
  #undef RLYPIN
  #define RLYPIN -1 //disable as pin 12 is used by analog LEDs
#endif

//automatically uses the right driver method for each platform
#ifdef ARDUINO_ARCH_ESP32
 #ifdef USE_APA102
  #define PIXELMETHOD DotStarMethod
 #elif defined(USE_WS2801)
  #define PIXELMETHOD NeoWs2801Method
 #elif defined(USE_LPD8806)
  #define PIXELMETHOD Lpd8806Method
 #else
  #define PIXELMETHOD NeoEsp32Rmt0Ws2812xMethod
 #endif
#else //esp8266
 //autoselect the right method depending on strip pin
 #ifdef USE_APA102
  #define PIXELMETHOD DotStarMethod
 #elif defined(USE_WS2801)
  #define PIXELMETHOD NeoWs2801Method
 #elif defined(USE_LPD8806)
  #define PIXELMETHOD Lpd8806Method
 #elif LEDPIN == 2
  #define PIXELMETHOD NeoEsp8266Uart1Ws2813Method //if you get an error here, try to change to NeoEsp8266UartWs2813Method or update Neopixelbus
 #elif LEDPIN == 3
  #define PIXELMETHOD NeoEsp8266Dma800KbpsMethod
 #else
  #define PIXELMETHOD NeoEsp8266BitBang800KbpsMethod
  #pragma message "Software BitBang will be used because of your selected LED pin. This may cause flicker. Use GPIO 2 or 3 for best results."
 #endif
#endif

//you can now change the color order in the web settings
#ifdef USE_APA102
 #define PIXELFEATURE3 DotStarBgrFeature
 #define PIXELFEATURE4 DotStarLbgrFeature
#elif defined(USE_LPD8806)
 #define PIXELFEATURE3 Lpd8806GrbFeature 
 #define PIXELFEATURE4 Lpd8806GrbFeature 
#else
 #define PIXELFEATURE3 NeoGrbFeature
 #define PIXELFEATURE4 NeoGrbwFeature
#endif

#include <NeoPixelBrightnessBus.h>

enum NeoPixelType
{
  NeoPixelType_None = 0,
  NeoPixelType_Grb  = 1,
  NeoPixelType_Grbw = 2,
  NeoPixelType_End  = 3
};

class NeoPixelWrapper
{
public:
  NeoPixelWrapper() :
    // initialize each member to null
    _pGrb(NULL),
    _pGrbw(NULL),
    _type(NeoPixelType_None)
  {

  }

  ~NeoPixelWrapper()
  {
    cleanup();
  }

  void Begin(NeoPixelType type, uint16_t countPixels)
  {
    cleanup();
    _type = type;

    switch (_type)
    {
      case NeoPixelType_Grb:
      #if defined(USE_APA102) || defined(USE_WS2801) || defined(USE_LPD8806)
        _pGrb = new NeoPixelBrightnessBus<PIXELFEATURE3,PIXELMETHOD>(countPixels, CLKPIN, DATAPIN);
      #else
        _pGrb = new NeoPixelBrightnessBus<PIXELFEATURE3,PIXELMETHOD>(countPixels, LEDPIN);
      #endif
        _pGrb->Begin();
      break;

      case NeoPixelType_Grbw:
      #if defined(USE_APA102) || defined(USE_WS2801) || defined(USE_LPD8806)
        _pGrbw = new NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>(countPixels, CLKPIN, DATAPIN);
      #else
        _pGrbw = new NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>(countPixels, LEDPIN);
      #endif
        _pGrbw->Begin();
      break;
    }

    #ifdef WLED_USE_ANALOG_LEDS 
      #ifdef ARDUINO_ARCH_ESP32
        ledcSetup(0, 5000, 8);
        ledcAttachPin(RPIN, 0);
        ledcSetup(1, 5000, 8);
        ledcAttachPin(GPIN, 1);
        ledcSetup(2, 5000, 8);        
        ledcAttachPin(BPIN, 2);
        if(_type == NeoPixelType_Grbw) 
        {
          ledcSetup(3, 5000, 8);        
          ledcAttachPin(WPIN, 3);
          #ifdef WLED_USE_5CH_LEDS
            ledcSetup(4, 5000, 8);        
            ledcAttachPin(W2PIN, 4);
          #endif
        }
      #else  // ESP8266
        //init PWM pins
        pinMode(RPIN, OUTPUT);
        pinMode(GPIN, OUTPUT);
        pinMode(BPIN, OUTPUT); 
        if(_type == NeoPixelType_Grbw) 
        {
          pinMode(WPIN, OUTPUT); 
          #ifdef WLED_USE_5CH_LEDS
            pinMode(W2PIN, OUTPUT);
          #endif
        }
        analogWriteRange(255);  //same range as one RGB channel
        analogWriteFreq(880);   //PWM frequency proven as good for LEDs
      #endif 
    #endif
  }
`

tonyn0, 2021-09-29:
Analog support came from my prodding/nagging. ;) I have not seen any issues so far. Please describe in detail something that we can try to duplicate.

9ghost9, 2021-10-15:
Hi . Sorry I have not gotten back to this thread as WLED version 12 seems to solve this issue using the Wemos D1 esp8266 module .

In the future I will post videos for each module and each version of WLED . Each video will clearly demonstrate what I see in testing  .

It is fair enough if you are eager to close this thread , but that will not render this query invalid .

Thank you .

9ghost9, 2021-12-02:
Had time to video ver. 12.0 which demonstrates perfect working order ->

https://www.youtube.com/watch?v=rMbqVQzwGAM

Aircoookie, 2021-12-02:
Hey, did you retry this with 0.13.0-b5? Inversion works correctly from my testing.

blazoncek, 2021-12-02:
I am at a loss. Why did you record a perfectly working set-up and not the one causing troubles?
I can see the same behaviour on 0.13 since b4 on every of my tests.
Could you record a misbehaving set-up?

9ghost9, 2021-12-02:
Ok had time to test latest ver. on Esp32 D1 .  Color order is perfect now . There is very limited range on the brightness slider.

9ghost9, 2021-12-02:
https://www.youtube.com/watch?v=rtctO7GQlaA

blazoncek, 2021-12-02:
This is odd as I am unable to reproduce. I am using IRLZ44N MOSFETs to drive 12V analog LEDs and not IGBT.
There is only 10k resistor pulling G low and G being attached directly to ESP32 GPIO pin.
Please check if something changes if you unselect gamma crrrection.
Also try the set-up with MOSFET.

blazoncek, 2021-12-02:
Please also check segment settings if there is brightness limited there.

9ghost9, 2021-12-02:
Mosfet is a-ok no issue .
Optocoupler - issues .
Currently testing Espixelstick 4.04 Beta firmware also for same optocoupler/igbt circuit  with same ESP32 D1 mini .
No issues with Optocoupler there  , just waiting for inverted fix .

For now Esp8266 D1 mini and WLED 12.0 is still best @ 100% working for Optocoupler .

Thank you for your time and effort . WLED is awesome.

blazoncek, 2021-12-02:
esp8266 uses 880Hz for PWM where esp32 uses 19kHz
Maybe this is the issue? Optocoupler not being fast enough?

9ghost9, 2021-12-02:
There is no issue with opto and pwm using Espixelstick firmware on the esp32.

9ghost9, 2021-12-02:
Wait a second , this is a slider/range issue not khz or hz issue .

blazoncek, 2021-12-02:
Esppixelstick may use different PWM frequency than WLED.
If MOSFET is working as expected then this proves that this is not an WLED issue since WLED knows nothing about what drives LEDs/lights. It outputs the same PWM signal.
As for the slider, you can open browsers developer tools and observe values submitted from browser to ESP. Those will be in range 0-255 so it is not a slider issue unless you are using a customised UI which will sumit other values.

9ghost9, 2021-12-02:
I had thought the 0-255 value was mapped down from 0-1023 ?

9ghost9, 2021-12-02:
You could be right about opto steps vs  pwm frequency . I will order faster opto to test this theory .
This example may prove you correct . 
Thank you for your suggestion .

https://www.analogictips.com/selecting-optocoupler-isolate-pwm/

blazoncek, 2021-12-02:
You can try compiling WLED from source and change PWM frequency in `const.h`

9ghost9, 2021-12-03:
Unfortunately there is a month of Thursdays worth in errors to compile the latest source .

9ghost9, 2021-12-05:
> You can try compiling WLED from source and change PWM frequency in `const.h`
blazoncek ,  You were 100 % correct !!!!

I edited PWM  as you suggested and loaded WLED into Platformio  .

All is right with the World . The slider now has full dimming range . 

Thank you very much .

Please close .

blazoncek, 2021-12-05:
Can you post the details of your changes so others can benefit?
Community would be grateful.

Please specify which IGBT you use, which optocoupler and what was the frequency that worked.
Thank you.

9ghost9, 2021-12-05:
Yes . as you suggested in const.h I change to this ->
This is for full slider range in WLED to work  with a 4n35 optocoupler .
I didn't test other frequencies as I knew the ESP8266 uses approximately  1k.
Igbt's need to be selected with values to suit your mains voltage . 
Due to safety concerns with AC power I cannot elaborate/suggest in this open forum .

I will answer questions privately on this forum on an as Diy/own risk basis.

`// PWM settings
#ifndef WLED_PWM_FREQ
#ifdef ESP8266
  #define WLED_PWM_FREQ    880 //PWM frequency proven as good for LEDs
#else
 // #define WLED_PWM_FREQ  19531
  #define WLED_PWM_FREQ  1000
#endif
#endif`

closed 2021-12-05 by Aircoookie