wled/WLED #2208: Cold white LED never turns on in RGBWC

hamed-ta, 2021-09-19:
When I set up Wled with my own custom esp8266 hardware which has 5 PWM channels for RGBWC, the cold white channel never turns on when I chose the cold white color.

WLED version 0.12.0

you can see in the pictures 

[image]
[image]
[image]

blazoncek, 2021-09-19:
Control of WW/CW is only supported via Alexa control.
We are working on full support.

KoffeinKaio, 2021-11-16:
I just hacked my chia RGBWC light from an WSB3L to an ESP8266 and ESP32 after that
With both current stable and current beta theres no way to control the C channel of RGBWC - cant seem to control it with alexa either.

Any way to get this to work? Or anyone knows a FW that supports this at the moment?
I dont need fancy effects, I just need an alexa controllable RGBWC light.

blazoncek, 2021-11-16:
> Any way to get this to work? Or anyone knows a FW that supports this at the moment? I dont need fancy effects, I just need an alexa controllable RGBWC light.

You can compile _cct-support_ branch or use pre-compiled binaries from @srg74 from my fork (experimetal).
But AFAIK Alexa should be able to control your RGBWW strip with current binaries.

blazoncek, 2021-12-18:
Fixed with CCT implementation.

closed 2021-12-18 by blazoncek