wled/WLED #2215: Error while compiling PIR usermod

Chillkroete1206, 2021-09-19:
Hi there,

I am trying to compile my own WLED binary for Wemos D1 mini with PIR-sensor support following [this](https://github.com/Aircoookie/WLED/wiki/Compiling-WLED) and [this](https://github.com/Aircoookie/WLED/tree/v0.11.0/usermods/PIR_sensor_switch) Documentation

Compiling via Visual Studio and PlatformIO results in the attached error. [Attached in a file due to backticks in the code, so if I would include it in my post, the backticks interrupted the quote]

Can anybody help me with this error or maybe provide a compiled binary for HC-SR501 PIR sensor running with a Wemos D1 mini?

Thanks in advance!

blazoncek, 2021-09-19:
You only need to:
```
  -D USERMOD_PIRSWITCH
  -D PIR_SENSOR_PIN=16 # (D0) on D1mini
```
In your platformio_override.ini env.
You can choose another pin of course.

From your error log it seems something wrong with your download. Use git to clone recent code.

Chillkroete1206, 2021-09-19:
Thanks for your reply. Which file in particular are you referring to?
 
I have several files named platformio_override.ini in various usermods folders (not in PIR_sensor_switch folder).
Further there is one platformio_override.ini.sample file in the root (WLED) folder.

blazoncek, 2021-09-20:
platformio_override.ini goes into wled folder

Chillkroete1206, 2021-09-20:
Hi,

my platformio_override.ini is included in the WLED-folder (not WLED00) and looks like this: [platformio_override.ini](https://github.com/Aircoookie/WLED/files/7197283/platformio_override.ini.txt).
I just adjusted the board I am using and added the above mentioned commands.

Unfortunately an [error keeps occuring ](https://github.com/Aircoookie/WLED/files/7197353/Error2.txt).

Any ideas for the reason? Thanks for your assistance!

blazoncek, 2021-09-20:
The error has nothing to do with platformio_override.ini but with the file being compiled.
Redownload source code.

Chillkroete1206, 2021-09-20:
Thanks for your persistent support! You were right. I screwed up the usermod_PIR_sensor_switch.h on my previous attemts. Sorry for avoidable mistake.

After fixing the issue, I am not sure, if your advice with the modified platformio_override.ini is still required. So I tried both, but none of them went without error.

Errors:
[Clean](https://github.com/Aircoookie/WLED/files/7198578/Clean.txt)
[Modified platformio_override.ini.txt](https://github.com/Aircoookie/WLED/files/7198581/Modified.platformio_override.ini.txt)

As you might have noticed earlier, I have absolutely no idea how to interprete the errors...

blazoncek, 2021-09-21:
With platformio_override.ini you create your compile overrides that are not lost when you download a new version from GitHub.
You have access to those if you click the _ant_/_alien_ icon on the left icon bar.

Upon first run you will often get errors, try to compile multiple times.
And check YT video from Quindor for custom compiling WLED.

Chillkroete1206, 2021-09-23:
Im half done with the video. No solution discovered yet, but I keep trying.

Would meanwhile anybody be so kind and upload a compiled bin-file for the following setup?
- Wemos D1 mini
- PIR_SENSOR_PIN=16 is fine
I am highly interested in understanding my issue(s), but I next weekend I would like to install my setup with the PIR sensor.
Your support is highly appreciated! Thank you so much!

blazoncek, 2021-09-23:
Please join Discord for help and support.
@srg74 is kind enough to provide pe-built binaries wit different usermods in his Wemos shield repo.

blazoncek, 2021-09-29:
Closing, not a bug in code.

closed 2021-09-29 by blazoncek