wled/WLED #2216: Animated_Staircase with 14 steps

Chillkroete1206, 2021-09-19:
Hi there,

I am intended to light my stairs with wled and found the [Animated_Staircase ](https://github.com/Aircoookie/WLED/tree/master/usermods/Animated_Staircase) usermod.
As I understood I need to segment each step. As far as I know segments are limited to 10 for an ESP32 and even less for an ESP8266.
However I have 14 steps, so is there a way to use this in my case?

Thanks and regards,

blazoncek, 2021-09-20:
Next beta will include 16 for 8266 and 32 for ESP32.

huggy-d1, 2021-09-22:
Have you considered this less capable but functional usermod: https://github.com/Aircoookie/WLED/tree/master/usermods/stairway_wipe_basic

blazoncek, 2021-09-22:
New beta with 16 segments for 8266 is out. Closing.

closed 2021-09-22 by blazoncek