wled/WLED #2220: Wipe effect on preset not starting at pixel 0

orxaz, 2021-09-21:
Hi.

In first place, many thanks for this amazing piece of software.

I would like to do a playlist to acomplish a similar effect that one of this video of the stairs.
I did a preset that turns on my stip with FX 3 (wipe). The problem I am having is that some times the effect begins at some point on the strip that is not the first pixel, sometimes at center, others almost at end, is random.

Is there a way to force the effect to start at first pixel?

My preset is like this:
{"seg":{"fx":3,"sx":230,"ix":127,"rev":true,"col":[[255,255,255],[0,0,0],[0,0,0]]}}

Sorry if this question was made before, I have not been able to find it.

Many thanks!

pushpinder107, 2021-09-25:
You can use the animated staircase usermod: https://github.com/Aircoookie/WLED/tree/master/usermods/Animated_Staircase

Aircoookie, 2021-09-25:
Hi!

There is a way to force the effect to start from the beginning:
Just add `"tb":0` to your preset, e.g. `{"tb":0,"seg":{"fx":3,"sx":230,"ix":127,"rev":true,"col":[[255,255,255],[0,0,0],[0,0,0]]}}`

Please reopen if that does not work :)

closed 2021-09-25 by Aircoookie