wled/WLED #2223: Custom mapping problems

mase76, 2021-09-22:
Hi!
I have a matrix 32x32 build with 4 matrices 32x8. I did a custom mapping in xLights, that worked fine.
To be more flexible, I wanted to use the custom mapping in WLED. I copied the mapping from xLights, and decremented every number, because, xLights starts with 1.
But some lines are horizontally displaced one LED.
Here is my ledmap.json. Maybe you can tell me, what's wrong.
`{"map":[
1016,1015,1000,999,984,983,968,967,952,951,936,935,920,919,904,903,888,887,872,871,856,855,840,839,824,823,808,807,792,791,776,775,
1017,1014,1001,998,985,982,969,966,953,950,937,934,921,918,905,902,889,886,873,870,857,854,841,838,825,822,809,806,793,790,777,774,
1018,1013,1002,997,986,981,970,965,954,949,938,933,922,917,906,901,890,885,874,869,858,853,842,837,826,821,810,805,794,789,778,773,
1019,1012,1003,996,987,980,971,964,955,948,939,932,923,916,907,900,891,884,875,868,859,852,843,836,827,820,811,804,795,788,779,772,
1020,1011,1004,995,988,979,972,963,956,947,940,931,924,915,908,899,892,883,876,867,860,851,844,835,828,819,812,803,796,787,780,771,
1021,1010,1005,994,989,978,973,962,957,946,941,930,925,914,909,898,893,882,877,866,861,850,845,834,829,818,813,802,797,786,781,770,
1022,1009,1006,993,990,977,974,961,958,945,942,929,926,913,910,897,894,881,878,865,862,849,846,833,830,817,814,801,798,785,782,769,
1023,1008,1007,992,991,976,975,960,959,944,943,928,927,912,911,896,895,880,879,864,863,848,847,832,831,816,815,800,799,784,783,768,
760,759,744,743,728,727,712,711,696,695,680,679,664,663,648,647,632,631,616,615,600,599,584,583,568,567,552,551,536,535,520,519,
761,758,745,742,729,726,713,710,697,694,681,678,665,662,649,646,633,630,617,614,601,598,585,582,569,566,553,550,537,534,521,518,
762,757,746,741,730,725,714,709,698,693,682,677,666,661,650,645,634,629,618,613,602,597,586,581,570,565,554,549,538,533,522,517,
763,756,747,740,731,724,715,708,699,692,683,676,667,660,651,644,635,628,619,612,603,596,587,580,571,564,555,548,539,532,523,516,
764,755,748,739,732,723,716,707,700,691,684,675,668,659,652,643,636,627,620,611,604,595,588,579,572,563,556,547,540,531,524,515,
765,754,749,738,733,722,717,706,701,690,685,674,669,658,653,642,637,626,621,610,605,594,589,578,573,562,557,546,541,530,525,514,
766,753,750,737,734,721,718,705,702,689,686,673,670,657,654,641,638,625,622,609,606,593,590,577,574,561,558,545,542,529,526,513,
767,752,751,736,735,720,719,704,703,688,687,672,671,656,655,640,639,624,623,608,607,592,591,576,575,560,559,544,543,528,527,512,
504,503,488,487,472,471,456,455,440,439,424,423,408,407,392,391,376,375,360,359,344,343,328,327,312,311,296,295,280,279,264,263,
505,502,489,486,473,470,457,454,441,438,425,422,409,406,393,390,377,374,361,358,345,342,329,326,313,310,297,294,281,278,265,262,
506,501,490,485,474,469,458,453,442,437,426,421,410,405,394,389,378,373,362,357,346,341,330,325,314,309,298,293,282,277,266,261,
507,500,491,484,475,468,459,452,443,436,427,420,411,404,395,388,379,372,363,356,347,340,331,324,315,308,299,292,283,276,267,260,
508,499,492,483,476,467,460,451,444,435,428,419,412,403,396,387,380,371,364,355,348,339,332,323,316,307,300,291,284,275,268,259,
509,498,493,482,477,466,461,450,445,434,429,418,413,402,397,386,381,370,365,354,349,338,333,322,317,306,301,290,285,274,269,258,
510,497,494,481,478,465,462,449,446,433,430,417,414,401,398,385,382,369,366,353,350,337,334,321,318,305,302,289,286,273,270,257,
511,496,495,480,479,464,463,448,447,432,431,416,415,400,399,384,383,368,367,352,351,336,335,320,319,304,303,288,287,272,271,256,
248,247,232,231,216,215,200,199,184,183,168,167,152,151,136,135,120,119,104,103,88,87,72,71,56,55,40,39,24,23,8,7,
249,246,233,230,217,214,201,198,185,182,169,166,153,150,137,134,121,118,105,102,89,86,73,70,57,54,41,38,25,22,9,6,
250,245,234,229,218,213,202,197,186,181,170,165,154,149,138,133,122,117,106,101,90,85,74,69,58,53,42,37,26,21,10,5,
251,244,235,228,219,212,203,196,187,180,171,164,155,148,139,132,123,116,107,100,91,84,75,68,59,52,43,36,27,20,11,4,
252,243,236,227,220,211,204,195,188,179,172,163,156,147,140,131,124,115,108,99,92,83,76,67,60,51,44,35,28,19,12,3,
253,242,237,226,221,210,205,194,189,178,173,162,157,146,141,130,125,114,109,98,93,82,77,66,61,50,45,34,29,18,13,2,
254,241,238,225,222,209,206,193,190,177,174,161,158,145,142,129,126,113,110,97,94,81,78,65,62,49,46,33,30,17,14,1,
255,240,239,224,223,208,207,192,191,176,175,160,159,144,143,128,127,112,111,96,95,80,79,64,63,48,47,32,31,16,15,0
]}
`

mase76, 2021-10-02:
Now I use the alignment feature provided with the latest beta. But there is the same displacement effect.
For example the 32x32 Skull from xLights:
[image]

blazoncek, 2021-10-05:
Why not correcting those lines in ledmap.json?
Nobody said that xlights and wled have compatible ledmaps. 😉

mase76, 2021-10-05:
Because it would be wrong with WLED effects, when I change the mapping that way.

closed 2022-04-27 by stale[bot]