wled/WLED #2224: Show Hardware Model on Manual OTA Page

MrPaulAR, 2021-09-23:
**Describe the solution you'd like**
Display the hardware info (esp8266, esp32, etc) on the manual OTA page.  I'm lazy and simply forget to look at the info page.

huggy-d1, 2021-09-24:
I think that is a fantastic idea.  Would help idiots like me keep from trying to load the wrong firmware.

bphermansson, 2021-10-13:
Something like this? (But better looking of course :) )

[image]

MrPaulAR, 2021-10-13:
Works for me.

MrPaulAR, 2021-10-13:
I believe this has gotten way off topic.  If the feature is useless then so be it, lets just close it out.

I would do this myself and submit a pull request but my front-end (and C++) skills are not up to par so I gave up.  

My goal wasn't to get under anyone's skin here.

softhack007, 2023-01-06:
Implemented in 0.14.0-b1

closed 2023-01-06 by softhack007