wled/WLED #2225: How can I change the startup Sequenz from the Software?

Caiberdady, 2021-09-23:
Hello,

I'm new with WLED. I'm an electronic-guy and used to Raspberry, arduino and so on...
ItTried to go the easy way....
What do I have?
ESP8266, Powersupply 5V/8A, LED: WS2812B
Softwareversion (tested all from 09-13)
What do I want?
I want to put power on the  power supply and a Sequenze of Macros should start...
It seems to I make a mistake somewhere...
It doesn't matter what I put in the settings, it always starts with amber light..
I put in Software V09 the boot Macro 1 (before I put something in Macro1) => works only, when I tip on the Macro...
I put in Software V13 2 Makros, switch on the "preset cycle" with 1 and 2 in it, it doesn't work...
Could you please put an explaination for DAUs? (DAU=Dumbest Assumable User)  :) 
Thanks in advance!

blazoncek, 2021-09-23:
Save a preset, and set that preset to activate at startup.

Please use Discord for support.

Caiberdady, 2021-09-23:
Thanks blazoncek,

I Don't know what you mean with Discord!
I installed Version 13 b3 with "Erase Flash".
I created one preset with name "First", Quick Load Lable "1" on ID "1".
Can't find the place to activate at startup...
I tried to put in Settings => Time & Makros something but don't know how!
And I don't get this:
_Macros have moved!
Presets now also can be used as macros to save both JSON and HTTP API commands.
Just enter the preset id below! Use 0 for the default action instead of a preset_
Do I have to send a "code" by HTTP API?

blazoncek, 2021-09-23:
If you managed to open an issue I'm sure you will find Discord. ;)

Hint: A link is on the WLED Github frontpage.

sansillusion, 2021-09-23:
In settings->led settings, then set the load preset at boot and preset
number

Caiberdady, 2021-09-23:
blazoncek Very helpfull... ;(  At least I Manage to be a Member on this Group in Discord. For other people who are struggeling, this is an APP!
@Sansillusion: THAT was really Helpfull! I would have never searched for a setting for makros in the light tab!!! Thanks a lot, you're my HERO this night!

closed 2021-09-23 by Caiberdady