wled/WLED #2226: On current beta (13 b3) - Setting more than 8 segments via JSON API (trough MQTT) is not working

oxivanisher, 2021-09-23:
**Describe the bug**
On current beta (13 b3) - Setting more than 8 segments via JSON API (trough MQTT) is not working.

**To Reproduce**
Try to set more than 8 segments trough the JSON API trough MQTT.

**Expected behavior**
The new max number of Segments (16 on ESP8266, 32 on ESP32) should be supported.

**WLED version**
 - Board: Wemos ESP32
 - Version 13 b3
 - Format Binary

**Additional context**
I think this also did not work as expected before the improved segment support in the current beta. But since I was almost at the limit with the number of segments, and used 8266 then (version 12), I did not think much about it. But now I tried to use more segments (12 in total) and it does not really work. In the following picture, there should be 12 segments. the first 8 get the correct name, the next two the correct number of LEDs but not the name and the rest seems to be ignored or lost:
[image]
I can reproduce this on another WLED instance and also tried it with omitting the name, which did not change anything.
My guess would be, that the API or its parser is not correctly working with the number of segments. Probably somewhere hardcoded.
Also I can set the "missing" segments via the WEB UI which works, but they then also don't get the API calls (updating neither color, state, name and so on).

Thank you for your help!

blazoncek, 2021-09-23:
This could mean out of memory condition on ESP8266.

oxivanisher, 2021-09-23:
I replaced all ESP8266 with ESP32. Total number of LEDs is < 100. Tried to find something in the json.cpp, but I did not find anything. It seems to work with `getMaxSegments` which should use the correct value.

blazoncek, 2021-09-24:
Can you post your whole MQTT message?

oxivanisher, 2021-09-24:
> Can you post your whole MQTT message?
> Maximum size of MQTT message is 1024 bytes.

That was it! After a major rewrite of my node red function to send each segment on its own, it is working now. Since I did not find this limitation documented, I added it to https://github.com/Aircoookie/WLED/wiki/MQTT .

blazoncek, 2021-09-24:
Wiki is outdated and should not be used.
Please fork WLED-docs and make a PR.

closed 2021-09-24 by blazoncek