wled/WLED #2228: Turning off LEDs with Alexa routine turns all WLED  devices to off

catarrer, 2021-09-24:
Hi, I set up 3 devices with WLED. In Alexa I App I managed to bind them to different rooms (where they actually are). I set a routine that turn OSS the lights at 22:30.
Every time the wled device in the living room is turned off by the device, both the other devices in the kids rooms turn to off... I don't know why.
When im telling Alexa to turn on/off a single light or room, only the wled device which is meant, turn on/off. Only in routine Alexa switches ALL devices.
Two devices on 13 beta2 and on on 12.0

huggy-d1, 2021-09-24:
Do you have the Sync feature configured so the living room WLED device automatically updates the effect / palette / brightness?

You might want to come up with a way to disable the sync feature before turning off the LR WLED device.  After turning it back on, have it run a preset to enable sync feature again.

The "how" part of enabling and disabling the sync feature would require more information about your current level of home automation, tools you are already using, and so on.

catarrer, 2021-10-01:
Thanks for your reply and sorry for delayed answering.

I checked my settings and sync is disabled in all three devices. Turning a single device with or without voice command does only effect the requested device. Automatic state changes by routines effect evertine all three devices.

geforcefan, 2021-10-09:
I know this bug from pre 0.12.0 version which is described at #1544. Afaik this is fixed in 0.12.0.

closed 2022-04-27 by stale[bot]