wled/WLED #2231: Toggle on/off with ir.json does not work very well

mase76, 2021-09-26:
Hi!
I added a custom ir.json to the filesystem. However on/off does not work very well. I have to press the button multiple times to get a reaction. All other buttons work fine. When I add my code with #define IRCUSTOM_ONOFF into ic_codes.h, the reaction is fine. This is the on/off part of my json file:
"0xFF02FD": { "label": "Toggle on/off", "pos": "1x4", "cmd": "T=2", "rpt": true },

blazoncek, 2021-09-26:
I have noticed that too occasionally but have attributed this to the _repeat_ function of buttons.
If I hold the on/off toggle button for a bit too long is starts repeating (you can see 0xFFFFFFFF sent) and that may confuse JSON code.
@scottrbailey may need to take a look.

mase76, 2021-09-26:
I removed the rpt option, but this did not nake it better

blazoncek, 2021-09-27:
It is handling of repeatable codes that is an issue, not your JSON.
You will need to wait for Scott.

scottrbailey, 2021-09-28:
It seems to be rather an issue handling of any value of &T=[0,1,2]
Calling from HTTP request works fine as does changing to {"on": true}

scottrbailey, 2021-09-28:
Also works when routing through JSON api.
"cmd": {"win": "&T=2"}

mase76, 2021-09-28:
But why are there no issues when defining the on/off toggle in ir_codes.h?

scottrbailey, 2021-09-28:
I tracked down the issue. It was because colorUpdated() wasn't being called for HTTP commands.

blazoncek, 2021-10-03:
Fixed in latest code.

closed 2021-10-03 by blazoncek