wled/WLED #2232: WSP32-ETH01 static ip not working

OlJohnny, 2021-09-26:
**Describe the bug**
When using the v0.13.0-b2 ethernet binary, setting a static ip in the network section doesn't seem to do anything. (Even after restarting)

**To Reproduce**
1. Set a static ip in the network settings
2. Try to connect ...

**WLED version**
 - Board: WT32-ETH01 ESP32
 - Version 0.13.0-b2
 - Format binary with ethernet support

Thank you for your help!

pruwait, 2021-10-11:
WSP32-ETH01 dhcp client with ethernet not stable working too.
After downgrade to v12 my WSP32-ETH01 get IP from UTP.

OlJohnny, 2021-10-30:
> Try beta 3. There were some changes in Ethernet code sice beta 2.

I have just tried Beta 4 and it doesn't seem to work.
Things I have tried:
- in the wifi settings leave network name empty
- restart the WT32-ETH01

Aircoookie, 2021-11-04:
Have you set static gateway (to the IP of your router) and static gateway (to, most likely, 255.255.255.0)?

OlJohnny, 2021-11-07:
> Have you set static gateway (to the IP of your router) and static gateway (to, most likely, 255.255.255.0)?

Previously I haven't changed the default values.
After setting the static gateway and rebooting (the "save & connect" button didn't seem to do the trick) it now works!
So thanks for the help! 👍

But it seems to me, that WLED should respect a static ip even without a set custom gateway (e.g. in a setup where one wouldn't want WLED to communicate to the internet)

Aircoookie, 2021-11-08:
I just tested this and the (ESP8266) WiFi framework will fallback to DHCP if no static gateway is specified (even if I remove the check done in WLED).

I will clarify that one must set a static gateway IP for the static IP to work in the documentation.
If you do not want the device to be able to connect to the internet, setting the static IP as the gateway as well seems to be a functional workaround.

closed 2021-11-08 by Aircoookie