wled/WLED #2233: Some effects not functional/accurate on SK6812 RGBWW 60 pixel/m

HardTurtle, 2021-09-26:
Pride 2015 and Rainbow chase as well as other effects not functioning properly with SK6812 RGBWW 60 pixel/m. Function normally with 30/m. Low frame rate, slow movement, small number of illuminated pixels between large dark gaps.

60p/m strip, 0.13 b2 or b3

Effects work properly on 30p/m strip, even simultaneuosly.

Quin-dig quad v1 ESP32

Flicker is refresh rate issue, note slow movement vs. smooth with 30p/m.

https://user-images.githubusercontent.com/91421668/134823636-d88b70ff-2713-402d-9126-c000eb82aaea.MOV

https://user-images.githubusercontent.com/91421668/134823630-d19fd765-2eef-4ee9-8080-b2a6564b0f69.MOV

Thank you for your help!

blazoncek, 2021-10-06:
WLED does not know about pixel density.
An BTW videos do not play, please post them to YT.

blazoncek, 2021-10-09:
Please post your cfg.json and presets.json (which only includes presets with problematic effects).

9ghost9, 2021-11-02:
Has this been solved ? 
I have noticed that the effects seem to sequence based on the total number of pixels rather than the output pixel count .

For example using 2 outputs ->

total pixels = 200       

output #1 = 100 pixels  and start = 0
output #2 =100 pixels  and start =  x.. any # tested

Effects will not display correctly . 

If I set **total pixels** = 100  the effects play correctly on each output .

HardTurtle, 2021-11-02:
> Has this been solved ? I have noticed that the effects seem to sequence based on the total number of pixels rather than the output pixel count .
>
> For example using 2 outputs ->
>
> total pixels = 200 [...]

I “solved” the issue by reverting to the official release from beta version. No additional or changes to configuration required.

LocalNic, 2021-11-30:
I just comment on a observation with sk6812 in  #2342  maybe it is also related to this. 

In short I have done some testing with an sk6812 Strip and observed something like a register shift.
Using a WEMOS D1 Mini with the newest WLED release (0.13.0-b5)

blazoncek, 2021-11-30:
Effect speed depends on the _segment_ LENGTH (not density) in pixel count for some effects.
Putting two strips with same LED count (and same segment set-up) but different LED density will produce the same speed on both strips. Your eyes/brain may interpret it differently due to shorter strip length on a more dense strip.

closed 2021-11-30 by blazoncek