wled/WLED #2235: Sporadically color change

Steude1, 2021-09-27:
Greetings everyone,

I’m very new to the whole "Wled World" and got my first problem in my current Installation, though please don’t judge me for a eventually noob failure.
I would be very thankful for any help about this.
In this Installation I have 6x 2 meters Sk6812 12V Stripes, powered by six 12V 4.2A Nextec Power Supplies and right after the third Stripe some LEDs changes the Color randomly for a short time and changes back to adjusted Color.
The Installation had a short test this morning and I thought at this time everything was running fine, but on the second test this evening the Problem showed up.
The Controller I use is this one https://shop.codm.de/automation/pixel/30/wlan-pixel-controller-wled?c=18, with a 12V-5V voltage converter, the Gnd of everything is connected but actually I have no Capacitors in the circuit.

Maybe someone could help me to fix this.
Thanks in advance.

pbolduc, 2021-09-28:
Please follow up on discord and close this issue.  Normally the problem is related to ground / wires. But you will get better help on Discord than here.

https://discord.gg/KuqP7NE

Steude1, 2021-09-28:
Ok, thank you very much.

closed 2021-09-28 by Steude1