wled/WLED #2238: Really long run

ascottbell, 2021-09-28:
Hi all,
I'm doing my first whole house, permanent Xmas light installation and it's looking like more than 1500 LEDs. Not sure how to get the desired effect of a continuous line of effects with the limitation in WLED. I;m running an ESP32 and 12v LED strips. any help is greatly appreciated.

blazoncek, 2021-09-29:
Look into my and @pbolduc recent work.
Pre compiled bins are available from @srg74

mjreddy, 2021-09-29:
@ascottbell, have you checked out Xlights? I started with aLEDs and WLED not too long ago and added Xlights to my arsenal specifically for a permanent house installation. There are so many help videos and a highly active community of xlights users on facebook that can help you through problems. Of course, WLED is great and can even receive the live E131/DDP/Artnet output from xlights so you can model in xlights and output to WLED.

Eddiiie, 2021-12-10:
> Look into my and @pbolduc recent work.
Pre compiled bins are available from @srg74

What work?  Link?

blazoncek, 2021-12-10:
Please use [WLED forum](https://wled.discourse.group) or Discord (link on front page) for help and support questions.
Closing.

closed 2021-12-10 by blazoncek

pbolduc, 2021-12-10:
The recent work referenced was with blazokcek and we implemented network output using DDP so multiple WLED instances can act like a single instance.  It is in the latest beta.  I believe there are some outstanding issues we have not addressed like when the master WLED instance turns off.