wled/WLED #2240: White channel issues with SK6812 and WS2811 on different ports

mjreddy, 2021-09-29:
I am running sk6812WW's on port 2 and WS2811 pixel nodes on port 1 of a QuinDigQuad. I have the LED types set correctly and when I change the auto-calc white to "accurate" the WS2811 pixel nodes completely shutoff when I select white. Is this the expected behavior? I assumed the WLED would only "auto-calc" whites for strips that were RGBW capable as set in the LED type and pass the "legacy" RGB white to non-RGBW strips. 

I am running build 2106302 on a QuinLED-Dig-Quad (ESP32)

Thank you for your help!

blazoncek, 2021-10-11:
This is not a bug but intended behaviour.
Use Dual mode if you want to control RGB white and White independently.

mjreddy, 2021-10-11:
Got it- thanks for the reply and removing the label. So with Dual mode it will calculate the white value for my RGBW capable strings  and just pass RGB to other strings?

blazoncek, 2021-10-12:
Use Brighter for full auto W calculations and Dual or Legacy or Accurate with W channel set to 0 for auto calculation.
Brighter will try to set W channel to the lowest RGB value, as will Dual and Legacy.
Accurate will auto calculate W channel and then subtract that from RGB.

blazoncek, 2021-12-18:
This has been fixed in recent code that introduces per-strip logic to apply auto-white calculation only if strip is actually RGBW.
Closing.

closed 2021-12-18 by blazoncek