wled/WLED #2241: dead nodemcu board

sayeed205, 2021-09-30:
Greetings ,
This is my 1st time working with wled and esp8266.
I flashed WLED_0.12.0_ESP8266.bin(687kb) via esphome flashed on nodemcu v2 .
It was working fine for some time then suddenly all led stopped its effect so i resupplied power then nodemcu stopped working and now it's heating like hell.
I didn't find any thread about this.
Also this happened to my two nodemcu

blazoncek, 2021-09-30:
You can throw it away. The ESP.
Most likely short-circuited somewhere but also possible manufacturing flaw.
Had a few of those.

sayeed205, 2021-09-30:
Same happened with two nodemcus
Sad :''''''(
My bad luck

sayeed205, 2021-09-30:
After flashing wled 
Vin should be as 5v
D4 should be as data in
And gnd and gnd 
And i was using 5v 2amp 10w power supply 
Is it okay ?

Or this could make short circuit ?

blazoncek, 2021-09-30:
I do not know. Never used NodeMCU.
Look at the technical data.
For support questions please use Discord.
Closing.

closed 2021-09-30 by blazoncek