wled/WLED #2242: WiFi initialisation takes percedence over Ethernet

blazoncek, 2021-09-30:
I have noticed, on several occasions, that Ethernet may not get connected if WiFi is configured.

Debugging the issue I noticed that Ethernet is properly initialised but then (after) WiFi gets connected and overrides Ethernet connection.
Web interface no longer responds to Ethernet IP but UDP broadcasts are still transmitted using Ethernet.

The issue appears if WiFi takes long to connect.

huggy-d1, 2021-09-30:
Sounds like a pesky bug to fix.  Thanks for uncovering the truth.

blazoncek, 2021-10-03:
Getting weirder!
Happened also without WiFi configured.
WLED AP fired up and no Ethernet connection available until I unplugged and re-plugged Ethernet cable.
@Aircoookie could you look into it?

Aircoookie, 2021-10-03:
This might be a difficult one. As far as I can tell, AsyncTCP (which is used by ESPAsyncWebserver) will bind the TCP socket to one specific interface when begin() is called. At that point, I'd expect the server to stop responding on the other interface, as it does not seem the code was designed for multiple network interfaces.

In case of Ethernet + WiFi, just inhibiting connecting to WiFi if an Ethernet type is configured is likely easiest (except if you can come up with a good use case for it), though I'd like Ethernet + AP to work.

blazoncek, 2021-10-04:
I will try to debug the issue a bit more since it occurs only sporadically. But when it does I can only resolve it by unplugging Ethernet cable.
It may take a while since network initialization code is a _spaghetti code_ to me. 😉

Aircoookie, 2021-10-04:
You're not the only one! Code gets amended all the time, at some point you reach a point where you can't see the forest for the trees. Some bug appears, a very obscure workaround gets implemented, and so on...
Currently in WLED i think connection management, LED state changed logic, and settings logic are the three points where we have arrived at this point the most. Settings will get resolved shortly 😉

blazoncek, 2022-01-10:
Fixed in https://github.com/Aircoookie/WLED/commit/04f5bdb84336062cf4e41db0e7d222f1a39147dc

closed 2022-01-10 by blazoncek