wled/WLED #2247: RGBW algorithm

harueg, 2021-10-03:
Hi
Thanks again for this great software !!!
I've just setup the ESP32 as an E1.31 controller with xlights.
In Xlights I've set the LEDs as RGB, in WLED as SK6812 RGBW, works great !
Only the white on LED chip gets to much dominance. My workaround for now is to set the saturation higher in xlights.
Would be great if there would be slider(s) to set the parameter(s) of the white algorithm. 

Well, yeah, saw that the fseq player is on the todo list, great !

Thanks

Harald

Aircoookie, 2021-10-03:
Hi!

By default, the Dual mode is selected. This means that, as long as the white slider is at 0, the white channel is auto-calculated using the "Brighter" algorithm. This sets the white value to the _lowest_ RGB value, approximately matching the saturation of the color.
There is also the "Accurate" mode, which will subtract from the RGB channels, so it's likely not helpful to you if you think there is too much white in Dual/Brighter mode, but you can give it a try regardless.
Lastly, "None" will just disable the white channel when receiving from xLights (and/or effects that use palettes).

Hope this helps at least a bit :)

harueg, 2021-10-04:
Hi Christian

Yes, I remember these settings from older versions of WLED. I am using the 0.13 b3. I don't see these options there.
Found it, wasn't that with each LED strand before ? well there was just one strand before.
There are even more options there, what can I expect from them ?

What is the meaning of palettes in this context ?

Yes, that did help !

Thanks  Harald

closed 2021-10-06 by harueg