wled/WLED #2248: Generic data pins for ESP32 and ESP32 eth

harueg, 2021-10-03:
Hi
What are the recommended data out pins. xlights did set one output pin on gpio12, which of course doesn't work.
I tried hard to find this info on the usual sites.

What ESP32 ethernet is referred to in the binary with esp32_eth ?
How many and which pins are left for LED data out ?

Thanks
Harald

blazoncek, 2021-10-03:
Any GPIO in ranges 0-5, 12-33 should normally work.
Ethernet does reserve quite a few GPIOs. Please refer to Discord pinned messages in #dev-general topic.

harueg, 2021-10-03:
I always had problems when connecting anything on gpio12, esp wouldn't boot.
But found a schematic of esp32_eth
[image]

Is the SN74AHCT a good choice or is the ADUM1400 better. Mr. Radig is using the ADUM1400 on his LED controllers, although a lot more expensive.

blazoncek, 2021-10-03:
GPIO12 is safe to use in any combination, pulled low or pulled high or left floating.
WT32-ETH01 may be buggy though, there were complaints...
I've ever only used 74HCT or 74AHCT or even 74LS for that matter. 😃

harueg, 2021-10-03:
Buggy on the hardwareside, or library ? Any other recommendation ?
For the pins I always used this : https://randomnerdtutorials.com/esp32-pinout-reference-gpios/
Is there a single chip solution from 3.3V data to 5V RS485 ? For long distance transmission.
I've used 3.3V RS485 over 12m and worked perfectly, but just to be on the safe side.
I think the ADUM1400 produces less spikes, due to the coil design.

Thanks  Harald

harueg, 2021-10-03:
There are various olimex boards : https://www.olimex.com/Products/IoT/ESP32/
some even have SD card slots, for playing fseq files.

blazoncek, 2021-10-03:
I use [this](https://www.tindie.com/products/twilightlord/twilightlord-esp32-ethernet-shield/) one.
And [this](https://quinled.info/quinled-esp32-ethernet/) one is also good.

harueg, 2021-10-03:
Thanks for the hint !
Also found this, but only has 4 outputs
https://www.tindie.com/products/k7bbr/wled-wt32-ethernet-controller-and-level-shifter/

blazoncek, 2021-10-06:
If your question is answered, please close the issue.

closed 2021-10-06 by harueg

BlackBird77, 2022-09-30:
I use pins GPIO 32 and 33 for APA102 Leds in Olimex Board..