wled/WLED #2252: 16 Bit Color Support

Differentunic, 2021-10-05:
I have a couple hundred UCS8904 leds, which are RGBW. I've tried to get them to work with WLED, but the colors see random for the whole length of leds. 

I've been experimenting with getting [FastLED](https://github.com/FastLED/FastLED) to work with RGBW, but the same problem with random colors ocoured. I searched around for datasheets on the UCS8904 Chip, and found [this](https://www.advateklights.com/knowledge-base/ucs8904) site, which shows that they use 16 bits for color.
<html>
<body>

lock Type | Data-Only
-- | --
Color Resolution | 16 bits
Physical Package | SOP8
RGB | See UCS8903
RGBW | Yes
Output Pixel Voltage | 5 – 24V
PWM Rate | 1000Hz
Suitable Camera | Up to 34fps, see Refresh Rate and PWM Rate
Data Transfer Rate | 800kbps
Redundant Data Line | No

</body>
</html>

I'm not the best at programming, bit managed to modify the RGBW code for 16 bits of color, which can be found [here](https://github.com/Differentunic/FastLED/tree/master/UCS8904).

If it's possible would you be able to add a setting in WLED to enable 16 bit color? I'm happy to help with testing changes with the leds, and if need be, help with the programming aspect.

Regards,
Differentunic

blazoncek, 2021-10-05:
This is really not a request for WLED.
WLED uses NeoPixelBus by @makuna so you should look there if the chip is supported and request support if not.
Once it is supported in NeoPixelBus we may include it in WLED.

Differentunic, 2021-10-05:
> This is really not a request for WLED.
> 
> WLED uses NeoPixelBus by @makuna so you should look there if the chip is supported and request support if not.
> 
> Once it is supported in NeoPixelBus we may include it in WLED.

Will do. I'll keep you posted on the progress of this. 👍

Regards,
Differentunic

huggy-d1, 2021-10-05:
A quick search of the NeoPixelbus repo for UCS8904 yielded 0 results.

Makuna, 2021-10-05:
Goto Github: Makuna\NeoPixelBus and create an issue, find a spec sheet and link it, and I can look into it.  The linked spec sheet isn't a spec sheet, its just a marketing sheet.  Doesn't give a clue about pulse timing.

NeoPixelBus already supports Rgb48Color (16 bits per channel color object), but there isn't a Rgbw64Color (Rgbw version).

Differentunic, 2021-11-04:
I have no idea how to add a new led type. I did try to just replace the SK6812 with the new UCS8904Feature, but there were errors in the compilation. (this was after updating the libs)

I do actually plan to use WLED for a Christmas setup but can't figure out how to add the 16-bit support into WLED. 

I'd really appreciate it if someone pointed me in the right direction of how to add this support, or wanted to collaborate to make it work. Really looking forward to it!

Regards

Aircoookie, 2021-11-05:
I too believe 16 bit output is a great step forward - even with only scaling the brightness this will give a noticable benefit to gradient smoothness at low brightness even if we stay with an 8 bit color model in the effect calculation for the time being.

One possible cause for the compilation error you had @Differentunic is that `NeoPixelBrightnessBus` (which is used by WLED) does not (yet) support the new types, the regular `NeoPixelBus` should though. BrightnessBus is kind of a compromise, since it has to scale the brightness on the fly and returning the original color is lossy. Still, it saves some memory compared to the alternative, double buffering. We could try switching to the regular `NeoPixelBus` object in `bus_wrapper.h` for testing.

Differentunic, 2021-11-05:
> We could try switching to the regular `NeoPixelBus` object in `bus_wrapper.h` for testing.

Is there some documentation surrounding how I would go about replacing it with the regular NeoPixelBus, or are you able to do it?

Makuna, 2021-11-05:
Note, there is an example in NeoPixelBus on how to implement a brightness bus using a shader technique.  The NeoPixelBrightnessBus was provided for backward compatibility to Adafruits; but I always suggest the shader as it give the sketch writer more control on how brightness is applied.

blazoncek, 2023-02-04:
It will be possible to use WLED with UCS890x chips with the (next) release of NeoPixelBus v2.7.3
See https://github.com/Makuna/NeoPixelBus/issues/638

Many thanks go to @Makuna for fixing NeoPixelBrightnessBus.

blazoncek, 2023-08-14:
Even though WLED does not natively support 16-bit color depth, it can utilise 16-bit LED pixels using on the fly up-scaling. Some 16-bit LEDs were successfully added as supported in 0.14.

Closing as it may not be feasible to update core to work  with 16-bit color depth.
Feel free to reopen if you find good justification otherwise.

closed 2023-08-14 by blazoncek

chefdeit, 2026-09-29:
> When does 16-bit add something to color output? Never, your eyes will not perceive any difference using 8-bit or 16-bit.

Here's an 8 bit PNG, whose background RGB value is 0,128,0, with some shapes whose RGB values are 0,127,0 and 0,129,0
[image]
People's vision can differ a little (or a lot), as well as their displays and hardware driving them. However, I can discern this 1 bit difference at a glance, and so were a few other folks to whom I showed this.

> Feel free to reopen if you find good justification otherwise.

The incredible power and value of WLED is in controlling multiple lights. The DMX support takes this to another level. 16 bit support will allow better color matching of the different LEDs and fixtures. Color mismatch cheapens the look. With 8 bits at low brightness levels this is particularly challenging.

Thank you!

blazoncek, 2026-09-30:
Now animate that picture and glance again. ;)

On a side note, WLED supports HUB75 displays with several thousands of pixels. Try telling ESP to make room in RAM for 4 channel 16-bit pixels in such case. I doubt you will be able to persuade it @chefdeit 

And BTW I am no longer associated with WLED.

chefdeit, 2026-09-30:
A big **Thank You**, @blazoncek Blaz, for replying so expeditiously, despite not even being with the project!

> Now animate that picture and glance again. ;)

I think if I make the text blink it'll pop even more :) Point being, turns out we can see between 8 bits / adjacent 256 levels, particularly in the green channel.

> On a side note, WLED supports HUB75 displays with several thousands of pixels.

Of course for any given color depth, including 8 bit, there's going to be one or more displays where all the pixels no longer fit on a given memory. Evolution of hardware. Fortunately, it happens on the memory side as well (e.g. ESP32-P4) for when an application calls for it, which with HUB75 displays it might.

For color match, at a minimum just green and red channels at 12 bits, if one had to squeeze things (perhaps at the cost of bandwidth / compute), would at least be able to match the 8-bit blue such that the overall color across multiple fixtures can still be calibrated to match even at lower output levels. Or any other similar tricks from image compression and early video card architectures - this is just the simplest one I know. Not a replacement for full 16 bit color across the board, but as an additional mode can perhaps get many more folks 70% there.

> And BTW I am no longer associated with WLED.

Would you mind tagging someone appropriate or re-opening this issue, then, so it's on the radar? And thank you so much for all your contributions, Blaz.

Alex | Chef de IT

DedeHai, 2026-09-30:
16bit native color is just not feasible, sorry. It would mean rewriting all color handling, that is a major endevour with very little usecases

blazoncek, 2026-09-30:
Not to mention CPU processing resources, usermods, integrations and other external systems.

IMO (as a photographer who dealt with image processing for several years) there is *no benefit* WLED would see from 16 (or 12) bit processing. It doesn't do color/hue shifts or process photographic images where increased color resolution would reduce posterization and other artefacts.

If one sees difference between green of 127, 128 and 129 the easiest fix is dithering on the boundary. As for above image, yes I can see square and "Can't you see?" letters on my color calibrated display but that doesn't mean you (or I) would be able to see a difference on actual WS2812 LED (which is itself only 8-bit).

And yes, I did add 16-bit *output* support for @Differentunic as he sent me his UCS8904 from down under. And while I was experimenting with it I didn't see any difference between values 127<<8 and 128<<8 or 32639.

As for low level brightness resolution, see my latest commits in my fork for chips that utilize variable constant current drivers (like APA102/SK9822/HD108/TM1814/SM16825).