wled/WLED #2256: running wled effects on some segments while controling different segments over DMX or using two different DMX modes (MultiRGB & SingleRGB) modes on different segments at the same time.

dzelionis, 2021-10-07:
My problem is that some software (SoundSwitch) i am using is not smart enought to control individual LED's (limited to 2 DMX universes) and i have not to many options use anything else if I want to get synhronization in between my DJ console (Denon Prime 4) and lightining...in general soundswitch does good job...in my case, i have 1200+ LEDs - that means i need ~7 DMX universes just for leds...if i want to control it all with a soundswitch (2 universes max) - my smart leds is not so smart anymore....
so i ended up controling it with Jinx, but again jinx has no integration to my Dj console....
  
Solution to that would be controling some stuff (some segments) from Jinx or LedFx or WLED effects where you need some more smart paterns,
and where not - control it with soundswitch in Single RGB DMX mode.

Is any plans to implement such feature ? 

Thanks
Best regards,
Domas Zelionis

dlandguth, 2022-02-24:
hello dzelionis. I stumbled upon your post, I'm investigating the feasibility of using Wifi enabled dimmer nodes (my friend built that run on WLED) with soundswitch to sync it with DJ board. Sounds like this is what you are doing? Could you point me in the right direction? Also, I'm using a mac if that makes a difference. Thanks for any help!
-Dave

dzelionis, 2022-02-25:
Hey @dlandguth,
I am still not there...where i want/plan to be...my setup bit extensive:
I have soundswitch connedcted to Denon Prime 4, and sound switch is good up to some level, as an example for not smart led strips i am using 30 channel DMX LED strip controller/dimmer which works very well with a sound switch. 
But i also  have 600 smart leds which i want to control using wled effects but it's needs to be synched to beat.
And  1600 smart leds which i still would use wled as a controller but, I would use it in DMX mode and data would come from Resolume...
Also I DIY 20 Watt RGB Laser Projector, which is controled from Pangolin Beyond....
so, as you see threre a big list of the components, which has to work  as symphony...i mean synchronyze color pallete in between, has to react to beat, have a correct bpm set, etc..... 

So i have some python scripts  (I am not an expert in C) in place which, to do a job, but thats still far away from where i am planning to be....so from python i am talking to wled via API, then to Resolume also Rest API, to pangolin Beyond via Integrated tcp server (pango scripting language), to sound switch via midi....but again, i was not happy about sound switch midi clock, so i have tried ableton link  to Prime4, but that again has some issues and it does not fit my needs....so no i am looking to get StageLinq from Prime4 reverse engineered for beatiinformation...so we have some progress on it with guys who has made StageLinq node inmplementation for pulling  StateData from DenonDj...so if we get that righ i should know all the detail about loops, bpms, phases, clock etc....so then you can play around starting different effects or lets say restarting wled's effect everytime track is looping.....

Some demnos of my setup:
https://drive.google.com/file/d/1IiUDrycsQZ4XKwqKvGImfXbyEi0491HS/view?usp=sharing
https://drive.google.com/file/d/1orWqpMoSVFoxnpn3TRQuVlFdUjgfPy_e/view?usp=sharing
https://drive.google.com/file/d/1dzs1198dpLIlDDmns8wLTTGk_hxDQIcK/view?usp=sharing
https://drive.google.com/file/d/1Uwshm6yLsdCF_7-Y9i2HKmaw7ADD-gJ5/view?usp=sharing
https://drive.google.com/file/d/1UnAAiPXNKuHs2Ms2BHAMrU8DLOuz79Ef/view?usp=sharing

I guess, thats not  what your was looking for, as i assume you lookng for somehting ready out of box, isn't?
and my python scripts its still in the dev stage where i would not be sharing with anyone just yet...as its messy and it would be shame to show it to a publics..needs alot of cleaning before i could publish it.....and i have reall life and work....so time wise i am limited too.....So, thats my story....some day i hope to get to the end of it....but not tomorrow....

dzelionis, 2022-02-25:
But if your question is how to make wled to be discovered and usable by Soundswitch  using ArtNet DMX...i have work arround here too, wrote  a python script wich will advertise it via Artnet, so basicaly you can run it on any machine on the lan, and it will pretend to be wled and advertise/broadcast it to soundswitch....

dlandguth, 2022-02-25:
Hello, thank you for the reply -- WOW WHAT A SET UP!! I hope to see it at Burningman one year!! I wish you lived in Los Angeles... I love the laser out the window too! My partner just got a pulse laser and its been so much fun!

My setup is a bit simpler...and I haven't really jumped into all this stuff yet technically. My techie partner (datadrian) created some WLED nodes we use to control WRGB LED strip made into large globe lights and diamonds. These are portable and easy to take to the beach to light it up for beach parties! See video below.

I'm looking to now level up and use soundswitch to control the lights so that I can script light shows to individual music tracks. What I imagine is just treating each of the 10 lights like a 4 channel wash light (this is how soundswitch would see it, as a wash fixture). Is there a program I can run that receives DMX from soundswitch and then pushes that out to the nodes via wifi? Yes I would love if there is something ready to go I can implement right away -- but if there is not I will have to learn more about programing...hard to get time to do that currently but one of these days.

Video I cut together of several parties:
https://youtu.be/r42HA0_8dB0

dzelionis, 2022-02-25:
Hey, thats nice... i like beach parties too! :D 
anyway, far as i know wled's implementation of DMX is not full, it works if your lighting software is using broadcast address to send DMX/ARTNET packet, but in case of soundswitch it would do discovery first and then use unicast address to send a DMX/ARTNET  packets...Because WLED does not advertise itself, sound switch would not send anything....But in your case, to use unicast is good thing, because you use wifi, so its better not to flood everyone if there is no need....so to make it work i could share that script....which does the job...but it was never tested on the mac...i am running it on linux...and if you have soundswitch and that script on the same pc, you may need tweak few things...again not sure about mac, but in general script is binding to some standart tcp ARTNET port, so if you have only one netwoek interface on pc it could be an issue...i resolve that on windows if i have 2 physicval network cards or one physical and one logical (loopback) ...but you need to start script before sondswitch, as soundswitch is very aggresive, it would take all the ARTNET ports on all available interfaces....starting script after would lead to faield results....give me few days, i will clean it up abit and will put few comments....also it would do multiple advertisements, as i asume that each light you have is a wifi client with different ip, so you need to advertise every single one..
when i will have script ready, i will drop here shared link. i say i should have it in few days, max comming monday.

dlandguth, 2022-02-26:
On a deadline for work but WOW thanks so much, I would love to try your script and see if it could work for me! No rush at all if you want to make changes, it may be a week or two before I have time to really play with these things. Cheers!

dzelionis, 2022-02-26:
Ok..i have all ready for you....screenshot in action:
https://drive.google.com/file/d/1RHg99N5zxGfP3XwbttQhiZBnogViaVzK/view?usp=sharing
script it self:
https://drive.google.com/file/d/1vEflDgHrkKSaZrM-XEhOCNfWtrGHY2qe/view?usp=sharing

you need to modify few thing at very top of that script to suit your needs...i am running it on python3.9 with no issues....
I hope it will work out for you, give me some feedback after you try.....
Thanks
dzelionis

dlandguth, 2022-03-02:
dzelionis amazing!! Thank you so very much!! 

I have downloaded the script and I will try it soon after work project. I will definitely let you know how it goes and how I'm able to make it work!

Cheers!
-David

Elucker69, 2022-04-11:
[image]

So I have a fixture running Arduino and esp8266 that is Wi-Fi enabled that I currently control using the WLED app on my phone.  I have a Denon Prime 4, Soundswitch USB DMX dongle, and a Windows computer.  So I think I have all the hardware I need. After carefully reading through your posts I have a couple of questions.  I know next to nothing about programming so like I said I'm over my head here.  Do I just run the script in Windows or do I need to download python?  And then what do I use to control / program the LED strips?

dzelionis, 2022-04-11:
> 

Hey,
yes, you need to install python3.x (3.7 or 3.8 or 3.9) would do a job...you can install using microsoft store:
https://docs.microsoft.com/en-us/windows/python/beginners#:~:text=To%20install%20Python%20using%20the,from%20the%20results%20under%20Apps.

if you give me your networks details such as ip used on wled netmask our windows pc ip i will configure for you a python file correctly...

dzelionis, 2022-04-11:
[image]
[image]

Elucker69, 2022-04-11:
Thank you so much that is very generous of you.  I installed Soundswitch last week when I finally got the dongle and I am just learning the software. I installed Soundswitch on the desktop computer in my home lab/workshop but I realize now I need to get a laptop to make it a little more practical.  So, I will get in touch when I get myself a laptop and transfer the Soundswitch license.  Hopefully this weekend if not sooner.  I don't know how many led's are on the light.  I will have to take it down and count them. It will be amazing to tie that in with the rest of my gear! https://www.etsy.com/people/ChaseHannaCreations?ref=shop_home_header

dzelionis, 2022-04-11:
Np at all, let me know if you need more help, i could help you to configure python script acording your network setup...
Best,
Dzelionis

Elucker69, 2022-04-27:
OK. Got a laptop.  Installed Engine DJ, Soundswitch, Bluestacks and Wled, Python 3.9 and Script.  Now the newbie question: How do I "run the script"? Neither of my Developer friends speak Klingon, nor Python.  Might have time to get back to this for a day or two. Thanks!

dzelionis, 2022-04-27:
start cmd ( windows key + r) and enter "cmd"
that will start windows shell...
if python is installed correctly,
you should be able type "python3" + enter
that would start a python in shell....so now if you put in
print("helllo!")
it should print it if you are in python if you are still in windows shell
you would get an error for this command....

So after you validate that,
just run python3 "script_name.py"
don't forget to edit the script beforehand and fix network settings at the
very top  to match your network setup....

Let me know if you are still struggling....

Thanks
Dzelionis

]

erhan-, 2022-05-01:
Hey @dzelionis !

I am trying to use my WLED bulbs in my room as well with the soundswitch on my denon prime go. I am currently running in to some issues (https://github.com/diyhue/diyHue/issues/814) because I use a hue bridge emulator and was trying to control them like that until I found out about artnet.

So I am trying to run your script currently on a rpi and have following errors:

1) The line: 10x : pack_into("bbbb", packet, offset, *tuple(ipSet))
throws the following error as soon as the soundswitch connects:

´´´
struct.error: byte format requires -128 <= number <= 127
´´´

I saw that you have a commented out line above that:
pack_into("!HHHH", packet, offset, *tuple(ipSet))

with that it runs, the artnet device is also visible on the controller but the byte arrangement of the ip address seems wrong. The interface name shows up as Artnet, 0.192.0.168, 1

Do you know what might be the issue here?

Will you create a repository for your small project?

dzelionis, 2022-05-02:
@erhan- so, do you have it sorted ? or do you need some help? :D

erhan-, 2022-05-02:
Yep, everything works good with that settings set on both WLEDs. I have not installed the soundswitch on my computer. I want to test the trial version. I think I will when I receive the athom bulbs. I flashed the novostella bulbs with the 1m version and have no ota anymore. So I am stuck on that version and your workaround will be the permanent solution for those two bulbs I have :) If you want to, you can create a repo for it or I can do it if you want to.

dzelionis, 2022-05-02:
@erhan- , also we have managed to reverse engineer stageLinQ protocol fully, frew months ago got tat beatinfo part done. So thats would you could use for getting the correct bpm or any loop stuff/phase of the beat and etc.....as i need to get all properly sync to other applications....you can find more info here:
https://github.com/MarByteBeep/StageLinq/discussions/12#discussioncomment-2283047

Aircoookie, 2022-05-03:
Closing as this seems resolved :) A new option is now per-segment live mode, which would allow you to use Grouping - driving e.g. 5 LEDs as if one, and thus reduce the amount of universes required.

closed 2022-05-03 by Aircoookie

dzelionis, 2022-05-03:
> Closing as this seems resolved :) A new option is now per-segment live mode, which would allow you to use Grouping - driving e.g. 5 LEDs as if one, and thus reduce the amount of universes required.

Hey Aircoookie, that's great! would it allow to do groups of leds in "wled effect" dmx mode (11 dmx channel mode) ? as that would be really powerful thing to have.....
Thanks
Best regards
dzelionis

Aircoookie, 2022-05-03:
Sorry, the Effect DMX mode sets the selected segment(s) only. To set individual LEDs (or groups of LEDs), you'll need the Multi RGB or Dimmer + Multi RGB mode.

dzelionis, 2022-05-03:
@Aircoookie , did you have any plans to get that implemented, as I think, it would bring Wled to the very top of all other DMX/Artnet controllers on the market. Myself, i am semi-fine, as again, did some python to get the same result, so python controls segments in Effect DMX mode, and i control python as multiple fixtures....but that's a workaround....

Elucker69, 2022-05-14:
@dzelionis, After stumblig around like a toddler in the dark I have made some progress.  I have run the script successfully however it is not finding the hardware.   I do not understand how to set up the network settings.  I  have both the internal wifi card and a usb wifi adapter.  Can you guide me how to modify the script to work with my settings? Thanks.

Elucker69, 2022-05-14:
@dzelionis In one of your screen shots it looks like your computer is listed in the hardware section of Soundswitch dmx & Art-Net list.  Mine is not.  Is there another step that I am missing?

blazoncek, 2022-05-15:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions as issue has been closed an is not intended for a discussion.
Locking the issue.