wled/WLED #2271: TM1814 RGBW -> In "solid" mode, color randomly goes red for a sec

cmpGI, 2021-10-14:
My build:
Wemos D1 Mini Esp8266 with LevelShifter TXS0102
To connect everything: 3way cable 0,75mm each line

Behavior:
TM1814 led strip randomly goes red for a sec, no matter what solid color I use. Multimetr show voltage around 11.30 at the end of the strip for 12v and ground, for data around 5.40V (not solid, jumping from 5.4x to 5.5x)
I tried different wled version from 0.12.0 to 0.13.0. I built 3 different boards, used different TM1814 strips which weren't longer than 5m - still the same issue.

**WLED version**
 - Board: Wemos D1 Mini ESP8266
 - Version: 0.12.0 -> 0.13.0-b4
 - Format: Binary

AgentHawk, 2021-12-15:
did you solve your problem?

cmpGI, 2021-12-15:
> did you solve your problem?

Barely but yes... had to re-soldered everything with thicker cables (between wemos - level shifter - led strip) afer that it goes stable.

blazoncek, 2021-12-15:
TXS0102 may be too slow in some instances.
Try using 74AHCT125 or similar.

AgentHawk, 2021-12-15:
same problem with a QuinLED DigUno V3

cmpGI, 2021-12-15:
> TXS0102 may be too slow in some instances. Try using 74AHCT125 or similar.

I will order few of them and give a try - thanks

blazoncek, 2021-12-15:
GPIO pins used?

cmpGI, 2021-12-15:
>

cmpGI, 2021-12-15:
> GPIO pins used?

GPIO 2 as data output which is connected to the level shifter

closed 2021-12-15 by cmpGI