wled/WLED #2272: WLED_0.13.0-b4: Issue adding segments to setup

rce1086, 2021-10-14:
ESP8266 + 2 strings of 300 WS2811 LEDS, currently connected as 1x600 on GPIO2
Please see attached - trying to add a second segment to treat this as 300 + 300 in 2 segments for effects purposes
[image]

Clicking the **tick** button behaves as though I've clicked **remove,** (not even visible) producing a pop-up error "You have to have multiple segments to delete one"
This doesn't happen on an instance that had segments created on v12.0, saved, and them upgraded...

blazoncek, 2021-10-15:
LED settings?

blazoncek, 2021-10-16:
Is this a new install or an upgraded one?
Try clearing your browser cache & local storage then retry.

rce1086, 2021-10-16:
This is a new install. My first one is the 12-release => 13-beta that works.
- I mostly use the WLED app on iOS, but use the PC for the reports; will try…
- 
thanks, Carl

blazoncek, 2021-10-16:
Cannot reproduce.
Try erasing flash and re-uploading firmware.

rce1086, 2021-10-16:
OK, thanks for trying…
I’ll do this, and report back.

rce1086, 2021-11-09:
Hi, I’ve just set up a brand new ESP32 to control my Christmas lights, and I see exactly the same behaviour from 0.13.0-b4:
LED-Config has 4 outputs (d15,d2,d4,d5); I ticked “create a segment per output”
Go to segments tab, try to set grouping to 3: it worked for segment 0, but deleted seg.1 when I clicked *tick*….
My current workaround is to flash 0.12, create the segments, save them to preset-16, then flash back to 0.13.0-b4

rce1086, 2021-11-09:
I think I’ve found the cause - it behaves if I have the UI configured for segments by start & stop LED, **but** it *doesn’t work* if it’s configured for segments by start & length!
I’ve flipped this back & forth a couple of times, with consistent results 😀

blazoncek, 2021-11-09:
Start tracing JSON messages from browser to wled (websockets or HTTP requests) and post the one that happens after you click save checkmark.

rce1086, 2021-12-28:
Please find attached a packet dump of the click on the tick icon, which as before, results in the segment being deleted.
This was recorded with b6, BTW
[WLED-seg-deletion-issue01.txt](https://github.com/Aircoookie/WLED/files/7782249/WLED-seg-deletion-issue01.txt)

blazoncek, 2021-12-28:
There is no need for wireshark traces (they jus add clutter and this is not a network issue).
Use your browsers developer tools and open network tab to see clean data exchange between browser and ESP. You may need to look into WS response tab.

blazoncek, 2022-04-27:
Closing due to 0.13.1 being released.
If the issue persists please reopen.

closed 2022-04-27 by blazoncek