wled/WLED #2274: How to at compile time Disable automatic brightness limiter？

zx804262156, 2021-10-15:
How to at compile time Disable automatic brightness limiter？

Aircoookie, 2022-04-16:
Add this build flag to your environment: `-D ABL_MILLIAMPS_DEFAULT=65000`

closed 2022-04-16 by Aircoookie

SpeakingOfBrad, 2022-04-29:
> Add this build flag to your environment: `-D ABL_MILLIAMPS_DEFAULT=65000`

This doesn't actually uncheck the "Enable automatic brightness limiter" checkbox, right? Just sets it so high that it won't go into effect?

Aircoookie, 2022-05-03:
True. I should add that `-D ABL_MILLIAMPS_DEFAULT=0` turns it off entirely.

ryansabin, 2022-09-30:
Setting this flag `-D ABL_MILLIAMPS_DEFAULT=0` fails to compile.

From wled_eeprom.cpp line 271 setting milliampsPerLed to 0 disables ABL. How would I set this to 0 at compile.
strip.milliampsPerLed = 0; //disable ABL

blazoncek, 2022-10-01:
> Setting this flag `-D ABL_MILLIAMPS_DEFAULT=0` fails to compile.
> 
> From wled_eeprom.cpp line 271 setting milliampsPerLed to 0 disables ABL. How would I set this to 0 at compile. strip.milliampsPerLed = 0; //disable ABL

It does for me.

processor83, 2023-11-11:
it compile but when you go to led config page and press save it give error because 0 is not a correct value and the checkmark stay there. I really need top disable ABL ar compile time any solution?