wled/WLED #2275: Preset related mapping

Sturm2002, 2021-10-15:
**Describe the solution you'd like**
Having multiple mapping files, or configs which can be combined with a Preset, by having another "Combobox " in the Preset settings, where you can select a mapping-config

**Additional context**
For Example, mapping-config1 is combined with Preset 1 Let's say it's from top to bottom
mapping-config2 is combined with Preset 2, and is just from the middle to the edges.

**Why?**
Would allow to for example safe "Pictures" for a LED Matrix by changing the mapping, but it would be too much work to update the mapping config everytime, and it wouldn't be possible to change it by a Time related Preset, or for example in a playlist

Thank you for maintaining this wonderful Project <3

Sturm2002, 2021-10-15:
): ok thank you anyways :D

blazoncek, 2021-10-15:
I will keep this issue open nevertheless. As a reminder. 😉

Sturm2002, 2022-01-21:
Hey, 
wouldn't it be possible to achieve that feature by creating a 2-Dim "mapping" array ?
i.e.

{"map":
    [0, 1, 2, 3, 4, 9, 8, 7, 6, 5, 10, 11, 12, 13, 14, 19, 18, 17, 16, 15, 20, 21, 22, 23, 24, 29, 28, 27, 26, 25],
    [4, 3, 2, 1, 0, 5, 6, 7, 8, 9, 14, 13, 12, 11, 10, ....],
    [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, ...]
}

and then link map[0] with preset[0], map[1] with preset[1] and so on, or make it configurable

blazoncek, 2022-01-21:
no, this will severely limit ledmaps.

blazoncek, 2022-08-18:
Added in #2737

closed 2022-08-18 by blazoncek