wled/WLED #2278: New Wifi 6 router, cannot join my network?

bwag62, 2021-10-16:
I'm no pro at this by any stretch and I waited weeks before posting because it seems so trivial but I need help. Several months ago & replaced our home router with a TPLink Mesh Deco Wifi 6 set up. Every device in my home was able to connect but my lights. I updated the new SSID & password on WLED but it will not connect and my router doesn't see the the light either. Please let me what info you may need to assist....MUCH APPRECIATED!!

toto79, 2021-10-17:
if you can't configure seperate SSID for 2.4 and 5 GHz, try to disable 5GHz (at least for a test), because ESPxxxx are not designed for 5 GHz, so this could lead to initial "handshake" issues with some routers.

bwag62, 2021-10-19:
Cannot configure separate SSID for 2.4 & 5, i can just toggle them on & off. I'm currently trying to find out what channel the 2.4 is, I'm not sure where to look. I have to admit all these router settings & options with 3 & 4 letter abbreviation mean nothing to me, sorry I'm apparently over my head on this one.  Thanks