wled/WLED #2280: Add ESP32-CAM board support

darranwil, 2021-10-17:
**Is your feature request related to a problem? Please describe.**
Would like to see ESP32-CAM board added to supported boards. 

**Describe the solution you'd like**
Ability to compile and install on ESP32-CAM boards. Cheapest ESP32 I found with external antenna connector. Remove camera module and SD support since really not needed.

**Describe alternatives you've considered**
Tried with esp32dev, flashed fine but no wifi AP or serial output.

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

darranwil, 2021-10-17:
Added to platform.ini and recompiled/flashed. Saw AP for once for a second and then nothing.

[env:esp32cam]
platform = espressif32@2.0
; platform = espressif32
board = esp32cam
; board = esp32dev
; change microcontroller
board_build.mcu = esp32
; change MCU frequency
board_build.f_cpu = 240000000L
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp32} -D WLED_RELEASE_NAME=ESP32 #-D WLED_DISABLE_BROWNOUT_DET
lib_deps = ${esp32.lib_deps}

daniboyghub, 2022-06-20:
Hi guys, any news about this?
I have a esp32-cam and flashed with wled, but my lets don't turn on. Any idea?

softhack007, 2022-07-02:
ESP32-CAM should work when compiling with `esp32dev`. However the problem is the strange pinout - all pins are connected to builtin CAM, SDCard or Serial interface.

[image]

Pink mean "wired to onboard connections".

closed 2022-10-21 by Aircoookie